platformio/platform-espressif8266 #1: typo

sticilface, 2016-06-13:
should be 
```
platformio platforms install https://github.com/platformio/platform-espressif/archive/develop.zip
```

not 
```
platformio platform install https://github.com/platformio/platform-espressif/archive/develop.zip
```

even then the url doesn't work.

ivankravets, 2016-06-13:
Where did you find it? `platformio platforms` will be deprecated in PlatformIO 3.0. Need to use `platformio platform` instead. This repo is intended for PlatformIO 3.0

closed 2016-06-13 by ivankravets

sticilface, 2016-06-13:
https://github.com/platformio/platform-espressif on the readme

ivankravets, 2016-06-13:
Thanks. This is a valid command for PlatformIO 3.0. If you would like to try PlatformIO 3.0, please follow to https://github.com/platformio/platformio/issues/423

sticilface, 2016-06-13:
I'm trying to get platformio to build with the latest staging on travis... not having much luck though.  
can't get it to build locally either but different reason.

ivankravets, 2016-06-13:
PlatformIO 3.0 is located in separate branch (not in `develop`). See https://github.com/platformio/platformio/tree/feature/platformio-30

To install it use:
```
pip uninstall platformio
pip install -U https://github.com/platformio/platformio/archive/feature/platformio-30.zip
```

Don't forget to report about issues if you will find them.

sticilface, 2016-06-13:
I'll definitely try this... but I'm in the middle of an epic trying to get auto build deployment working with travis..