platformio/platform-espressif8266 #4: section '.text' will not fit in region 'iram1_0_seg'

profra, 2016-09-10:
What kind of issue is this?
[x ] PlatformIO Core. If you’ve found a bug, please provide an information below.

**I am not sure if it is bug... I have found this link esp8266/Arduino#1582 but I am not able to solve this problem in Platformio... Arduino IDE builds without any problem... Please help if possible**

_Configuration_
Operating system: W10

_PlatformIO Version (platformio --version)_:
 CLI 3.0.1, IDE 1.5.0 (Atom)

_Description of problem_
I cannot build simple programm with library....

_Steps to Reproduce_
pio run

_Actual Results_
... see the attached picture

_Expected Results_
... simple build

_If problems with PlatformIO Build System:_
_The content of platformio.ini:_
```
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
```
_Source file to reproduce issue:_
used library https://github.com/sumotoy/TFT_ILI9163C/tree/Pre-Release-1.0p7
main programm ... basicSetup.ino
```
#include <SPI.h>
#include <TFT_ILI9163C.h>

#define __CS  0   //(D3) ... fp
#define __DC  5   //(D1)
#define __RST 4   //(D2)
/*
 SCLK:D5
 MOSI:D7
*/

TFT_ILI9163C tft = TFT_ILI9163C(__CS, __DC, __RST);

void setup() {
  tft.begin();
  tft.print("Hello World!");
}

void loop(void) {
}
```
Additional info

[image]

ivankravets, 2016-09-14:
We have just created special `stage` branch that allows working with latest Arduino Framework with "1 click". See http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

ivankravets, 2016-09-14:
See comment https://community.platformio.org/t/platformio-3-0-broke-all-my-esp8266-projects/747/7?u=ivankravets

closed 2016-09-14 by ivankravets

n8henrie, 2016-09-15:
Worked brilliantly, thanks!

I didn't bother installing the dev version of platformio -- though this was recommended so may be necessary in some cases. My steps:

1. Remove all the includes / linking I had been trying as a workaround from `build_flags`
1. `platformio platform uninstall espressif8266`
1. `platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage`
1. Set `platform = espressif8266_stage` in `platformio.ini`

Thank you!

sticilface, 2016-09-15:
I had to still manually install the feature/stage in step 3, which seems to overwrite the normal platformio directory

unless i did this it said ```espressif8266_stage``` not found.  

I'm using the development version of platformio too...  I guess what i would be aiming for is to have the two versions side by side.. and to switch just using that platform flag.  so i could even build both at the same time. 
:)

ivankravets, 2016-09-15:
@sticilface both version should work. I've just checked it. Could you try again?

vlast3k, 2016-11-10:
i am still somehow facing problems with this..i uninstalled all esp stuff and then installed just the stage
```
admins-mini-2:vThingCO2 admin$ platformio platform uninstall espressif8266
Uninstalling espressif8266 @ 1.2.0:     [OK]
Uninstalling framework-arduinoespressif8266 @ 1.20300.1:        [OK]
The platform 'espressif8266' has been successfully uninstalled!
admins-mini-2:vThingCO2 admin$ platformio platform uninstall espressif8266_stage
Uninstalling espressif8266_stage @ 1.3.0-alpha1:        [OK]
Uninstalling framework-arduinoespressif8266 @ 1.20400.0-dev:    [OK]
Uninstalling sdk-esp8266 @ 1.10502.0:   [OK]
Uninstalling tool-espotapy @ 1.0.0:     [OK]
Uninstalling tool-esptool @ 1.409.0:    [OK]
Uninstalling toolchain-xtensa @ 1.40802.0:      [OK]
The platform 'espressif8266_stage' has been successfully uninstalled!
admins-mini-2:vThingCO2 admin$ platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage
PlatformManager: Installing platform-espressif8266
git version 1.9.3 (Apple Git-50)
Cloning into '/Users/admin/.platformio/platforms/installing-WPU0LS-package'...
remote: Counting objects: 92, done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 92 (delta 46), reused 51 (delta 21), pack-reused 0
Unpacking objects: 100% (92/92), done.
Checking connectivity... done.
espressif8266_stage @ 1.3.0-alpha1 has been successfully installed!
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Downloading...
Unpacking...
toolchain-xtensa @ 1.40802.0 has been successfully installed!
PackageManager: Installing tool-esptool @ ~1.409.0
Downloading...
Unpacking...
tool-esptool @ 1.409.0 has been successfully installed!
PackageManager: Installing tool-espotapy @ ~1.0.0
Downloading...
Unpacking...
tool-espotapy @ 1.0.0 has been successfully installed!
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has b
```
and then after building
```
/Users/admin/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/node
mcu/firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [.pioenvs/nodemcu/firmware.elf] Error 1
```

ivankravets, 2016-11-10:
@vlast3k you should specify new platform in project configurattion file, platformio.ini
```
platformio = espressif8266_stage
```

vlast3k, 2016-11-11:
ahhh it turned out that with the stage version.. apparently the new SDK the
code becomes larger
after stripping a lot of the code
with the stable version i get
text       data     bss     dec     hex filename
324332     5580   31144  361056   58260 .pioenvs\nodemcuv2\firmware.elf

and with stage
text       data     bss     dec     hex filename
353721     7576   31408  392705   5fe01 .pioenvs\nodemcuv2\firmware.elf

and before stripping the code - with stable it was
text       data     bss     dec     hex filename
391461     8548   31544  431553   695c1 .pioenvs\nodemcuv2\firmware.elf

i wasn't really expecting this and since it coincided with this topic that
was raised a month ago - i decided that this is the case - sorry for this

tjuerges, 2017-04-26:
Same problem.  Almost a tiny project.  Pristine installation of platformio:  same bug.  Following advice from @n8henrie  Success!

I really do not grok what the root of the issue is.  Can somebody please explain why the stage version can link binaries which fit into the board memory when the 'default' version cannot?  Thanks a lot!

n8henrie, 2017-05-05:
Sorry, this is all above my head to be honest -- I just got lucky with
trial and error. Glad it helped someone else though!

Nate