platformio/platform-espressif8266 #7: xtensa toolchain path not set properly

AtomikPunk, 2016-09-22:
Hi there, I'm sorry if that's not the right place to post...

I just installed platformio-ide to do some development with an esp8266, namely the WeMos D1 mini board using the arduino framework. However, the initialization step does not seem to properly set the xtensa toolchain path... Everytime I hit ctrl-s to save/build, an error message displays telling me that:
```
'xtensa-lx106-elf-g++' is not recognized as an internal or external command,
operable program or batch file.
'xtensa-lx106-elf-ar' is not recognized as an internal or external command,
operable program or batch file.
```
If I open a platformio terminal window from within the IDE (Atom) and manually set the path to the toolchain bin directory (located in %USERPROFILE%\.platformio\packages\toolchain-xtensa\bin since I am on Windows...), it does the trick, but for that terminal window only. I mean that, from that terminal window, I can do `platformio run` just fine and the build will run properly.

I don't know if its really a bug or just a misconfiguration, nor if it is only on windows...

ivankravets, 2016-09-22:
What to you see with `pio --version`? Do you have global PlatformIO CLI in your system?

AtomikPunk, 2016-09-23:
Hi Ivan, it shows `PlatformIO, version 3.1.0` and I did not install the CLI separately.

`pio platform list` in my project folder gives
```
espressif8266 ~ Espressif 8266
==============================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things
 applications.

Home: http://platformio.org/platforms/espressif8266
Packages: tool-mkspiffs, tool-esptool, framework-arduinoespressif8266, toolchain-xtensa
Version: 1.0.3
```

`pio update` gives
```
Platform Manager
================
Platform espressif8266
--------
Updating espressif8266 @ 1.0.3:         [Up-to-date]
Updating tool-mkspiffs @ 1.102.0:       [Up-to-date]
Updating tool-esptool @ 1.409.0:        [Up-to-date]
Updating framework-arduinoespressif8266 @ 1.20300.0:    [Up-to-date]
Updating toolchain-xtensa @ 1.40802.0:  [Up-to-date]

Library Manager
===============
```
Thanks!

ivankravets, 2016-09-23:
What is version of Windows OS?

AtomikPunk, 2016-09-23:
Windows 10 home

AtomikPunk, 2016-09-27:
I'm sorry I'm pretty new to PlatformIO... Can you point me to where the path(s) "should" be set (script? config file? another tool chain example?) so I could have a look and try to locate it myself? Thanks!

ivankravets, 2016-09-27:
We have thousands Windows users for PlatformIO IDE and you are the first with this issue. I can't imagine what is wrong with your OS. All should work from a box.

Could your provide TeamViewer session for debug?

AtomikPunk, 2016-09-27:
I'll try to uninstall/reinstall and see it the problems goes away. I'll keep you posted.

How does the framework/command line generally finds the various toolchains? Is it using environment variables (e.g. PATH) or the path is stored somewhere in some config files? Or something else...?

ivankravets, 2016-09-27:
PlatformIO appends toolchains on-the-fly, see https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/pioplatform.py#L73

Tell me please, do you see toolchain in `%USERPROFILE%/.platformio/packages/toolchain-xtensa`?

AtomikPunk, 2016-09-27:
Thanks for the python source link!

Yes, I can see the toolchain in my `%USERPROFILE%/.platformio/packages/toolchain-xtensa/bin` folder. Could that be that the package is misconfigured somehow?

This is the folder I manually add to my path so I can compile. As I said in my first post, if I manually add this path to my PATH environment variable, I am able to compile/link successfully. PlatformIO just does not seem to find it automagically...

ivankravets, 2016-09-27:
Could you provide TeamViewer session for debug (  me@ikravets.com )?
https://www.teamviewer.com

AtomikPunk, 2016-09-27:
Unfortunately no, I cannot... That would be quite complicated and I am not even in front of my development computer right now :/

AtomikPunk, 2016-09-28:
Thanks you very much for troubleshooting with me. I uninstalled and reinstalled the platformIO IDE, terminal and all the packages (toolchains, platforms, uploaders, etc...) and now it seems to work! I don't know which one did the trick but I suppose that, for some reason, my first install was corrupted... Thanks again!

closed 2016-09-28 by AtomikPunk