platformio/platform-espressif8266 #10: mkspiffs ignoring custom flash layout

stufisher, 2016-10-04:
I'm having a funny issue where mkspiffs is ignoring my custom defined flash layout. I'm working on an esp01 a like board with 1M flash and want to specify 64k spiffs so i can retain OTA updates.

Running platformio run --target uploadfs
results in it trying to upload a 256k spiffs.bin despite me having specified

```
[env:esp12e]
platform = espressif8266
framework = arduino
board = esp01_1m

upload_port = /dev/tty.SLAB_USBtoUART
build_flags = -Wl,-Tesp8266.flash.1m64.ld
```

running with verbose is showing:
```
$ platformio run -v --target uploadfs
[Tue Oct  4 20:40:06 2016] Processing esp12e (platform: espressif8266, build_flags: -Wl,-Tesp8266.flash.1m64.ld, upload_port: /dev/tty.SLAB_USBtoUART, board: esp01_1m, framework: arduino)
---------------------------------------------------------------------------------------------------------
"mkspiffs" -c data -p 256 -b 4096 -s 262144 .pioenvs/esp12e/spiffs.bin
/homie/ui_bundle.gz
```

-s 262144 would appear to be the wrong value.

I can make this work if i run mkspiffs manually, and specify the correct spiffs_start with esptool.

This has been working just fine up to now, so im not sure if this an issue in platformio 3.1 or a recent update to the espressif platform, or some configuration issue at my end?

ivankravets, 2016-10-04:
Could you show me an output of 
```
pio platform show espressif8266
```

stufisher, 2016-10-04:
```
pio platform show espressif8266
espressif8266 ~ Espressif 8266
==============================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Version: 1.1.1
Home: http://platformio.org/platforms/espressif8266
License: Apache-2.0
Frameworks: simba, arduino

Package toolchain-xtensa
------------------------
Type: toolchain
Requirements: ~1.40802.0
Installed: Yes
Description: xtensa-gcc
Url: https://github.com/jcmvbkbc/gcc-xtensa
Version: 1.40802.0

Package framework-simba
-----------------------
Type: framework
Requirements: >=7.0.0
Installed: No (optional)

Package tool-esptool
--------------------
Type: uploader
Requirements: ~1.409.0
Installed: Yes
Description: esptool-ck
Url: https://github.com/igrr/esptool-ck
Version: 1.409.0

Package tool-mkspiffs
---------------------
Type: uploader
Requirements: ~1.102.0
Installed: Yes
Description: Tool to build and unpack SPIFFS images
Url: https://github.com/igrr/mkspiffs
Version: 1.102.0

Package framework-arduinoespressif8266
--------------------------------------
Type: framework
Requirements: ~1.20300.1
Installed: Yes
Description: Arduino Wiring-based Framework (ESP8266 Core)
Url: https://github.com/esp8266/Arduino
Version: 1.20300.1

Package sdk-esp8266
-------------------
Requirements: ~1.10502.0
Installed: No (optional)
```

ivankravets, 2016-10-04:
This issue caused by an upcoming update for PlatformIO 3.2. I'll fix it

ivankravets, 2016-10-04:
Please run `pio platform update`

closed 2016-10-04 by ivankravets

stufisher, 2016-10-04:
Many thanks for your speedy response! All working again :)

commit d9ce3656 (2016-10-04): Handle user custom LD script for "buildfs" // Resolve #10
commit cb33c30d (2018-01-02): Merge branch 'develop' into feature/stage

* develop:
  Fix SPIFFS uploading in lazy mode
  Handle user custom LD script for "buildfs" // Resolve #10
  Hot fix for "uploadfs" target while PlatformIO 3.2 isn't released // Resolve #9
  Bump version to 1.1.0 (issue #6)
  Additional target "buildfs" to accompany "uploadfs" // Resolve #6
  Add new ESPrectro board
  Use stable documentation
  Fix flash size for BlynkBoard

# Conflicts:
#	platform.json