platformio/platform-espressif8266 #9: target uploadfs

gallochri, 2016-10-04:
After last update to espressif 1.1.0 uploadfs target fault:

`gallochri@mediacenter:~/dev/gallNFC> pio run --target uploadfs
[Tue Oct  4 13:06:42 2016] Processing esp12e (platform: espressif8266, lib_install: 28,63,64, upload_speed: 460800, board: d1_mini, framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-mkspiffs @ ~1.102.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-mkspiffs @ 1.102.0 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
TypeError: coercing to Unicode: need string or buffer, NoneType found:
File "/usr/lib/python2.7/site-packages/platformio/builder/main.py", line 134:
env.SConscript("$BUILD_SCRIPT")
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/home/gallochri/.platformio/platforms/espressif8266/builder/main.py", line 327:
join("$BUILD_DIR", "spiffs"), "$PROJECTDATA_DIR")
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 260:
return MethodWrapper.__call__(self, target, source, *args, **kw)
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Builder.py", line 626:
return self._execute(env, target, source, OverrideWarner(kw), ekw)
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Builder.py", line 547:
tlist, slist = self._create_nodes(env, target, source)
File "/home/gallochri/.platformio/packages/tool-scons/script/../engine/SCons/Builder.py", line 511:
target, source = self.emitter(target=tlist, source=slist, env=env)
File "/home/gallochri/.platformio/platforms/espressif8266/builder/main.py", line 161:
with open(env.GetActualLDScript()) as f:
========================================================================================================================= [ERROR] Took 0.84 seconds =========================================================================================================================`

ivankravets, 2016-10-04:
Thanks a lot for the report!

Please run `pio update`

closed 2016-10-04 by ivankravets

gallochri, 2016-10-04:
All ok now, thank you.

commit 498c52d9 (2016-10-04): Hot fix for "uploadfs" target while PlatformIO 3.2 isn't released // Resolve #9
commit cb33c30d (2018-01-02): Merge branch 'develop' into feature/stage

* develop:
  Fix SPIFFS uploading in lazy mode
  Handle user custom LD script for "buildfs" // Resolve #10
  Hot fix for "uploadfs" target while PlatformIO 3.2 isn't released // Resolve #9
  Bump version to 1.1.0 (issue #6)
  Additional target "buildfs" to accompany "uploadfs" // Resolve #6
  Add new ESPrectro board
  Use stable documentation
  Fix flash size for BlynkBoard

# Conflicts:
#	platform.json