platformio/platform-espressif8266 #12: function as parameter -> "was not declared in this scope"

stozk, 2016-11-16:
Hi all,

I have a problem with an Arduino Sketch for a NodeMCU (esp8266) board.

It compiles on the Arduino IDE and everything works fine, but when I try to build it in platform.io,
I get this error:

....: error: 'callback' was not declared in this scope

void setup() {
client.set_callback(callback);
}

void callback(const MQTT::Publish& pub) {
...
}

Any idea what might be causing this?
If I change the callback function params, it seems to accept the declaration but then an error is thrown regarding wrong function params.

BR
Daniel

ivankravets, 2018-01-03:
Please change extension of the file to INO or http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

closed 2018-01-03 by ivankravets

ArtiomLK, 2018-10-23:
If you move the callback method before it is invoked in the setup method the code should work. This [document](http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually) has more information about your issue.