platformio/platform-espressif8266 #15: Faster flashing for ESP8266 (and CC3200) boards

ivankravets, 2016-11-27:
# Moved from https://github.com/platformio/platformio/issues/592

/cc @georgkreimer, @Nzbuu, @Jorgen-VikingGod, @georgkreimer, @eni23 

-------

Flashing of the ESP8266 boards takes a while. Anybody who has flashed some of them noticed it's about a minute. I saw [this post](https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds) blog today. Those guys at Cesanta Software have created a [modified version of the esptool](https://github.com/cesanta/fnc/blob/master/common/platforms/esp8266/stubs/esptool.py) but also have another flash tool called [Flash'N'Chips](https://github.com/cesanta/fnc) which reduces the flash times significantly. It would be great to have the flash time reduced for PlatformIO using these 2 options.

jtwalters, 2016-12-14:
I have had some good luck with setting the upload speed in platformio.ini:

```
upload_speed = 921600
```

Takes about 10 seconds to build and upload.

ivankravets, 2019-05-14:
The latest version uses esptool.py

closed 2019-05-14 by ivankravets