platformio/platform-espressif8266 #16: ESP8266 RTOS SDK

ivankravets, 2016-11-27:
# Moved from https://github.com/platformio/platformio/issues/402

/cc @andriyadi
-------

I'm about to adopt ESP8266 native SDK, and since my use cases require parallelism, it would be great to adopt RTOS SDK. I've found this project, [ESP-OPEN-RTOS](https://github.com/SuperHouse/esp-open-rtos), somehow better than the official Espressif's.

Would it be possible to support this kind of SDK into Platformio?

Thanks in advance. Have a great holiday guys!

adikabintang, 2017-03-20:
Hi, is there any news about this? Actually I have tried [1] but I failed. After several hours investigating the problems, I think the latest version of ESP RTOS SDK [2] no longer fits with the provided python script in that branch. Currently, I am modifying the sming.py in order to work properly but unfortunately it still does not work. Has anybody recently tried SmingRTOS on PlatformIO and succeed?  

[1] https://github.com/platformio/platform-espressif8266/tree/feature/framework-sming 
[2] https://github.com/espressif/ESP8266_RTOS_SDK

ivankravets, 2017-06-22:
Implemented. See example
https://github.com/platformio/platform-espressif8266/tree/develop/examples/esp8266-rtos-sdk-blink

closed 2017-06-22 by ivankravets

WallaceWilliam, 2019-03-02:
try it https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio