platformio/platform-espressif8266 #21: Support HTTP Upload in a simple way

Bluebie, 2017-01-17:
I use PlatformIO to develop hobby projects with the espressive8266 platform / huzzah board. There are situations where the ArduinoOTA upload method is unusable\*, so i'd like PlatformIO to support http as a hex upload port.

Currently I have a shell script which contains:

```
platformio run
curl -v -F "update=@.pioenvs/huzzah/firmware.bin" http://doorbell-11a251.local/update
echo
```

and I use this to do uploads, instead of the cmd+u shortcut in these projects. The extra_script method was too confusing to add support with the weird way the form argument is specified to curl.

I think this would be generally useful for other platforms. I could imagine it being really useful for people modifying firmwares on network routers and things like that, which usually use a http form for firmware upgrades. It could also be useful for upgrading devices on different subnets or even over cellular link. I propose the following:

the standard uploader should be modified so when upload_port begins with http:// or https://, it invokes curl in verbose mode and does a http post as above. A `platformio.ini` config file working in this mode would look like:

```
[env:huzzah]
platform=espressif8266
framework=arduino
board=huzzah
upload_port=http://192.168.1.20/update
```

---
\* ArduinoOTA can become unusable if you're using UDP for other stuff. The Arduino UDP library isn't great at supporting multiple concurrent services on different ports.

ivankravets, 2017-01-24:
Which source code should be used in this case to handle that request?

Bluebie, 2017-01-24:
On the device side? Check out the example code in the ESP8266 Arduino core: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino

universam1, 2017-07-19:
I second that!

makiolo, 2017-07-28:
I need this!

bigbuglamer, 2018-06-29:
spiffs can't be uploaded this way :( At least for now.

joba-1, 2018-11-05:
would be really useful to me

joba-1, 2018-11-06:
A great tool just got better for me - found out this feature is already available!

Maybe this issue should be closed?

# Example 
This works for a NodeMCU board with ESP8266HTTPUpdateServer/WebUpdater firmware.

## platformio.ini
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
{all your other options here}
upload_port = {board_ip_or_(m)dns_name}/update
extra_scripts = upload_script.py
upload_protocol = custom
```

## upload_script.py
Place this file in same directory as platformio.ini.
```
Import("env")
env.Replace(UPLOADCMD="curl -v -F image=@$SOURCE $UPLOAD_PORT")
```

nospam2k, 2020-02-29:
> ## upload_script.py
> Place this file in same directory as platformio.ini.
> 
> ```
> Import("env")
> env.Replace(UPLOADCMD="curl -v -F image=@$SOURCE $UPLOAD_PORT")
> ```

I'm getting Update error: ERROR[10]: Magic byte is wrong, not 0xE9 trying to upload spiffs.bin. The firmware file works ok. The error is coming from curl but I can't find anything online to fix the problem.

nospam2k, 2020-03-01:
I found why I'm getting the error and want to offer some addition to joba-1's example script for uploading either firmware or SPIFFS filesystem. The key is replacing "image" with the appropriate GET input name (see man curl for more info). I'm checking for the existence of the global variable SPIFFS_BLOCK to determine which I am attempting to upload. Then you can use the tool bar upload option for firmware or the run other target > PIO Upload Filesystem Image to upload SPIFFS:

```
Import("env")
try: SPIFFS_BLOCK = env['SPIFFS_BLOCK']
except KeyError: SPIFFS_BLOCK = None
if SPIFFS_BLOCK is None:
    env.Replace(UPLOADCMD="curl -u <username>:<password> -F firmware=@$SOURCE $UPLOAD_PORT")
else:
    env.Replace(UPLOADCMD="curl -u <username>:<password> -F filesystem=@$SOURCE $UPLOAD_PORT")
print(env['UPLOADCMD'])
```