platformio/platform-espressif8266 #23: Please update ESP8266 Arduino framework soon

huming2207, 2017-02-03:
Hello there,

Please upgrade ESP8266 Arduino framework mirror soon, as the main upstream project repo (https://github.com/esp8266/Arduino) has fixed some bugs, updated some SDK APIs and it's useful for developments.

If you are not convenient to update, could you please tell me how to create my own mirror/fork so that I can use the newest code. Thanks.

Thanks and Regards,
Jackson

huming2207, 2017-02-03:
Update:

I've directly downloaded the arduino framework and patched into ``/.platformio/packages/framework-arduinoespressif8266``, but there are quite a lot of linking issues when linking the ``firmware.elf``.

ivankravets, 2017-02-03:
See http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

closed 2017-02-03 by ivankravets

huming2207, 2017-02-03:
Just checked, it works, updated and compiled successfully. Thanks mate.