platformio/platform-espressif8266 #26: More descriptive memory report

zgoda, 2017-03-07:
Sming framework uses more descriptive format for memory usage:

```
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8394|     916
    rodata|           ReadOnly Data (RAM)|    3FFE83A0|    3FFE8C74|    2260
       bss|      Uninitialized Data (RAM)|    3FFE8C78|    3FFF0440|   30664
      text|            Cached Code (IRAM)|    40100000|    40106F85|   28549
irom0_text|           Uncached Code (SPI)|    40201010|    40237153|  221507
Total Used RAM : 33840
Free RAM : 48080
Free IRam : 4237
```

This is done by script https://github.com/SmingHub/Sming/blob/develop/tools/memanalyzer.py that uses xtensa-lx106-elf-objdump instead of xtensa-lx106-elf-size, it provides more details, specially IRAM left where ISR code is located. It would be nice to have such detailed report after build finishes.

ivankravets, 2018-06-04:
Done in https://github.com/platformio/platformio-core/commit/bfc94d36e36bc9e2164734cf3894241db1a4831e

`pio upgrade --dev`
`pio run -v`

sticilface, 2018-06-13:
this is a great feature, I have a small comment on the report
```
DATA:    [====      ]  44.2% (used 36232 bytes from 81920 bytes)
PROGRAM: [=         ]  10.7% (used 450796 bytes from 4194304 bytes)
```

I'm assuming that the 4194304 comes from linker or something, however, it is misleading as the max sketch size is 1Mb as on the 4Mb module 1Mb sketch + 3Mb for SPIFFS.  However, I believe this goes to 512K with OTA as it needs space for both bins at once so it is 512 + 512 + 3Mb.  I think even the Arduino env doesn't get this right

```
Sketch uses 412828 bytes (39%) of program storage space. Maximum is 1044464 bytes.
```

ivankravets, 2018-06-14:
@sticilface we already implemented this auto-calculation for ESP32. Could you file a separate issue https://github.com/platformio/platform-espressif8266/issues? I'll take a look how is difficult to implement the same for ESP8266

sticilface, 2018-06-25:
sorry to reopen!  

Is there a chance that this fix has broken the SPIFFS upload tool.  As I now get the following error
```
Auto-detected: /dev/cu.Repleo-CH341-00005014
Uploading .pioenvs/serial/spiffs.bin
warning: serialport_set_baudrate: baud rate 460800 may not work
warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08
Uploading 3125248 bytes from .pioenvs/serial/spiffs.bin to flash at 0x01048576
warning: Invalid size/address, too large: 2fb000 1048576
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
error: espcomm_upload_mem failed
*** [uploadfs] Error 2
```

ivankravets, 2018-06-25:
It was fixed a few days. Please PIO update

closed 2018-06-25 by ivankravets