platformio/platform-espressif8266 #27: staging out of sync

sticilface, 2017-03-08:
Compiling with stage gives new compile error..
due to this commit esp8266/Arduino@93413ef

➜  piotest pio run -e stage

[Wed Mar  8 22:36:30 2017] Processing stage (build_flags: -Wl,-Tesp8266.flash.4m.ld, lib_deps: ArduinoJson, ESPAsyncTCP, ESPAsyncWebServer, board_f_cpu: 160000000L, platform: espressif8266_stage, upload_speed: 921600, board: esp12e, framework: arduino, board_flash_mode: qio, upload_port: /dev/cu.wchusbserial1460, board_f_flash: 80000000L)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting test.ino
Collected 32 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ArduinoJson> v5.8.3
|-- <ESPAsyncTCP> v1.0.1
|   |-- <ESP8266WiFi> v1.0
|-- <ESPAsyncWebServer>
|   |-- <ESPAsyncTCP> v1.0.1
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoJson> v5.8.3
|   |-- <Hash> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <Hash> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs/stage/src/test.ino.o
Archiving .pioenvs/stage/libFrameworkArduino.a
Indexing .pioenvs/stage/libFrameworkArduino.a
Compiling .pioenvs/stage/lib/ESP8266WiFi/ESP8266WiFiScan.o
Compiling .pioenvs/stage/lib/ESP8266WiFi/WiFiUdp.o
Compiling .pioenvs/stage/lib/ESPAsyncTCP_ID305/AsyncPrinter.o
Compiling .pioenvs/stage/lib/ESPAsyncTCP_ID305/ESPAsyncTCP.o
Compiling .pioenvs/stage/lib/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.o
Compiling .pioenvs/stage/lib/ESPAsyncTCP_ID305/SyncClient.o
Compiling .pioenvs/stage/lib/ESPAsyncTCP_ID305/tcp_axtls.o
Compiling .pioenvs/stage/lib/Hash/Hash.o
Compiling .pioenvs/stage/lib/Hash/sha1/sha1.o
/Users/amelvin/git/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp: In static member function 'static void ESP8266WiFiScanClass::_scanDone(void*, int)':
/Users/amelvin/git/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp:297:41: error: cannot convert 'bss_info::<anonymous struct>' to 'bss_info*' in assignment
for(bss_info* it = head; it; it = it->next, ++i)
^
/Users/amelvin/git/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp:305:45: error: cannot convert 'bss_info::<anonymous struct>' to 'bss_info*' in assignment
for(bss_info* it = head; it; it = it->next, ++i) {
^
*** [.pioenvs/stage/lib/ESP8266WiFi/ESP8266WiFiScan.o] Error 1
================================================================================ [ERROR] Took 2.51 seconds ================================================================================

======================================================================================== [SUMMARY] ========================================================================================
Environment default    [SKIP]
Environment stage      [ERROR]
================================================================================ [ERROR] Took 2.51 seconds ================================================================================

ivankravets, 2017-06-22:
Does it work now?

ivankravets, 2017-06-26:
Please reopen if you need a help

closed 2017-06-26 by ivankravets