platformio/platform-espressif8266 #29: ELF conversion and flashing is currently broken

tarelda, 2017-03-24:
I discovered that scripts doesn't properly support other linker maps than stock one.  According to https://espressif.com/en/support/explore/get-started/esp8266/getting-started-guide because I use modules with 4M memory I adjusted linker script to have irom section starting at 0x40210000. Thus converting should yield two files flashable at adresseses 0x00000 and 0x10000. As far I undestand scons build scripts build process is designed for creating two image files for adresseses 0x00000 and 0x40000. Am I right? How this can be fixed?

tarelda, 2017-03-25:
During further testing I found out that builtin linker maps works with bootloaders like rboot. Which is not even mentioned in the docs.

ivankravets, 2018-01-03:
Duplicate of https://github.com/platformio/platform-espressif8266/issues/22

closed 2018-01-03 by ivankravets