platformio/platform-espressif8266 #33: toolchain-xtensa version 1.40802.0 is broken for armv6

elijah-taylor, 2017-05-13:
The packaged toolchain at https://dl.bintray.com/platformio/dl-packages/toolchain-xtensa-linux_armv6l-1.40802.0.tar.gz (most recent revision for ~6 months) is broken for armv6.

Two issues I've encountered when trying to follow the "quick start" instructions at http://docs.platformio.org/en/latest/quickstart.html on raspberry pi:

1) the contents of bin/ and xtensa-lx106-elf/ contain broken links to relative paths "bin/<elf-file>"
2) xtensa-lx106-elf-c++ seems to be calling the system assembler |as| instead of the toolchain specific xtensa-lx106-elf-as

I worked around by dumping the contents of https://dl.bintray.com/platformio/dl-packages/toolchain-xtensa-linux_armv6l-2.tar.gz with fake package.json info into my packages directory.  This package seems to be correct and I could compile the blink sample.

vortigont, 2017-05-24:
I have no such problems. Just installed esp platform with pio and everything runs fine (raspbian)

vortigont, 2017-05-24:
Looks like you were right.
Trying to install from scratch (clean OS/ install pio from cmd) results downloading a package with broken symlinks.

Updating from some old setup to the same toolchain-xtensa ver 1.40802.0 seems to work fine, and I still can compile.

```
pio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.3.0          [Up-to-date]
Updating tool-mkspiffs                   @ 1.102.0        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating tool-espotapy                   @ 1.0.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
```

elijah-taylor, 2017-05-24:
If I had to guess I'm assuming the old version had the proper symlinks and the new version is unpacked on top of the old one in place.  If this is the case, I would argue this is the wrong way to update packages, and that they should be installed pristine w.r.t. previous releases (I would guess this is a general platformio packaging issue, not sure where to file this).

ivankravets, 2017-06-22:
@valeros please check this package

ivankravets, 2017-06-26:
Thanks a lot for the report! Fixed!

Please run `pio update` command, it should work now. Please confirm.

closed 2017-06-26 by ivankravets

vortigont, 2017-06-28:
it works fine, tnx.