platformio/platform-espressif8266 #34: Add Wemos d1 mini lite

ehendriks, 2017-05-19:
I like to use platformio with a Wemos d1 mini lite. 
https://wiki.wemos.cc/products:d1:d1_mini_lite#documentation

MFornander, 2017-06-03:
Agree

MFornander, 2017-06-03:
Figured it out.  Use this to as your platformio.ini file.  That last two are the needed ones for D1 Mini Lite:

[env:D1MiniLite]
platform = espressif8266
board = d1_mini
framework = arduino
board_f_cpu = 160000000L
upload_speed = 921600

build_flags = -Wl,-Tesp8266.flash.1m64.ld
board_flash_mode = dout

ehendriks, 2017-06-05:
Yes, it works like a charm!
No need to keep the Arduino IDE just for compilation.
Thanks a lot.

MFornander, 2017-06-09:
Added a Pull Request for this at:
https://github.com/platformio/platform-espressif8266/pull/38

ivankravets, 2018-01-03:
Added, the new release is coming soon.

closed 2018-01-03 by ivankravets