platformio/platform-espressif8266 #37: LD scripts fail

sticilface, 2017-06-05:
Just upgraded to `version PlatformIO, version 3.4.0b10`

Now get failure on building
```
Error: Could not find 'esp8266.flash.4m.ld' LD script in LDPATH '/Users/amelvin/.platformio/packages/framework-arduinoespressif8266@src-e093741c24b1f814060be0cac08f8093/tools/sdk/lib /Users/amelvin/.platformio/packages/framework-arduinoespressif8266@src-e093741c24b1f814060be0cac08f8093/tools/sdk/ld /Users/amelvin/.platformio/packages/framework-arduinoespressif8266@src-e093741c24b1f814060be0cac08f8093/tools/sdk/libc/xtensa-lx106-elf/lib /Users/amelvin/git/MySONOFF/.pioenvs/debug'
```

using this in PIO as board definition

```
[env:debugsonoff]

platform = espressif8266_stage
framework = arduino
board = esp12e
build_flags = -Wl,-Tesp8266.flash.4m.ld-DDEBUG_CODE=Serial -DMY_SETTINGS_FILE=\"/sonoff.json\"  
board_flash_mode = qio
board_f_cpu = 160000000L
board_f_flash = 80000000L
upload_port = 192.168.1.151 ; square
;upload_port = /dev/cu.Repleo*
;upload_speed = 460800
lib_deps = ArduinoJson, ESPAsyncTCP, ESPAsyncWebServer, AsyncMqttClient, Bounce2
```
 Changing to ```build_flags = -Wl,-Teagle.flash.4m.ld``` fixes the issue

TheOriginalMrWolf, 2017-06-13:
Good tip - thanks!

Unfortunately, that worked fine until I tried a SPIFFS build & upload:
```
Error: Could not find 'eagle.flash.4m.ld' LD script in LDPATH ''
```

ivankravets, 2017-06-22:
Resolved in https://github.com/platformio/platform-espressif8266/commit/bb761c98a9345491aa73b927069959476025b081

commit bb761c98 (2017-06-22): Keep support for old LD Scripts // Resolve #37
closed 2017-06-22 by ivankravets