platformio/platform-espressif8266 #40: Failures in esptool(1)

michaesc, 2017-06-15:
The actual esptool(1) may be defective, when used on a modern [Amica (AI-Thinker ESP-14) NodeMCU R2 platform](http://www.electrodragon.com/product/nodemcu-lua-amica-r2-esp8266-wifi-board/):

```
user@host:~/.platformio/packages/tool-esptool$ ./esptool -v
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>

user@host:~/.platformio/packages/tool-esptool$ ./esptool -vv -cd nodemcu -cb 115200 -cp /dev/ttyUSB0 -cf $HOME/platformio/esplatest/.pioenvs/nodemcuv2/firmware.bin
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 115200
        setting port from /dev/ttyUSB0 to /dev/ttyUSB0
        espcomm_upload_file
        espcomm_upload_mem
opening port /dev/ttyUSB0 at 115200
        tcgetattr
        tcsetattr
        serial open
opening bootloader
resetting board
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
resetting board
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
resetting board
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
user@host:~/.platformio/packages/tool-esptool$
```

Please give constructive advice, if this is a user error. Thanks!

valeros, 2017-06-15:
Hi @michaesc!
Could you please try to upload a basic sketch using Arduino IDE? Does it work correctly?

michaesc, 2017-06-15:
No need to persue this @valeros because it didn't occur to me that the hardware might be defective. At the current price, it's no woder that one of my NodeMCU controllers fails in the way I described in this bug report.

When I switch to another NodeMCU microcontroller of the same architecture it works fine, so I've concluded that there's no problem with esptool(1) or platformio and will close this bug report.

closed 2017-06-15 by michaesc