platformio/platform-espressif8266 #43: Add Debugging via GDB stub

universam1, 2017-07-19:
Debugging via GDB stub is available:
https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub

Would be a killer feature for PlatformIO!

ivankravets, 2017-07-19:
We tested that and it does not work properly :(

ivankravets, 2017-07-19:
We tested that and it does not work properly :(

Does it work for you?

universam1, 2017-07-20:
Here is a good introduction:
http://hackaday.com/2015/12/12/squash-your-esp-8266-bugs-with-esp-gdbstub/

Actually I was hoping of PIO, guess they should be able to get some support by Espressif likely

minkir014, 2019-06-05:
Any thing new about this issue?

augustozanellato, 2019-09-01:
Any news? 
I could try it this evening probably

AgainPsychoX, 2024-07-06:
I visited the topic myself and I come up with come notes and minimal project , if anyone is interested: https://github.com/AgainPsychoX/TestGDBStubForESP8266