platformio/platform-espressif8266 #46: unable to upload SPIFFS

Alasknnj, 2017-07-29:
After using espressif8266_stage instead of espressif8266, and upgrading Platformio from 3.0 to 3.4, I am unable to run ``platformio --target uploadfs`` successfully. The single output from the command is:
``Error: Could not find LD script``

``platformio run`` and ``platformio run -t upload`` are working.

sidoh, 2017-08-08:
What does your `platformio.ini` look like? Just found that overriding the load script with `build_flags` (e.g., `-Wl,-Tesp8266.flash.1m64.ld`) wasn't working. Removed that and it worked. Looks like changing `esp8266` to `eagle` will work (see [this](https://github.com/platformio/platform-espressif8266/commit/cd2c849a6846d1d747f496cd5cdf224cdbc4b2a8) commit).

abhisheksafui, 2017-08-08:
facing same issue esp12e.json already has eagle instead of esp8266

baoshi, 2017-08-11:
It seems like a bug in the platform files.
1. You need to create a folder "ldscripts" under "espiressif8266_stage", and copy the ".ld" files from "framework-arduinoespressif8266xxxxx/tools/sdk/ld" into ldscripts.
2. You need to modify each .ld file's "INCLUDE" section, remove "../ld/" at beginning.
Now buildfs and uploadfs should work.
However if you want to choose a flash configuration, you need to make your own board.json file. It seems the build_flags does not override ldscripts in the build-in board definitions.

sidoh, 2017-08-11:
It looks like ld scripts were intentionally removed because they're supplied elsewhere: (commit: https://github.com/platformio/platform-espressif8266/commit/cd2c849a6846d1d747f496cd5cdf224cdbc4b2a8). 

Just replace

```
-Wl,-Tesp8266.flash.1m64.ld
```

with

```
-Wl,-Teagle.flash.1m64.ld
```

and it should work.

ivankravets, 2017-08-11:
Fixed in https://github.com/platformio/platform-espressif8266/commit/5a8b3ec61497abab1a7879e2a13cf053576559dc

Please `pio update`

closed 2017-08-11 by ivankravets

universam1, 2017-09-15:
not working for me:
espressif8266_stage             @ d192958        [Up-to-date]
Core 3.5.0a10

pio run -t uploadfs
Processing main (framework: arduino; lib_deps: ArduinoJson, ESPAsyncTCP, ESPAsyncWebServer, JPEGDecoder
, 2; board_f_cpu: 160000000L; platform: espressif8266_stage; upload_speed: 921600; board: d1_mini; monitor_baud: 115200")
--------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Could not find LD script

nardev, 2018-01-17:
getting same error again... tried -Wl,-Teagle.flash.1m64.ld
and pio updates.. nothing helps.. any clue???

ivankravets, 2018-01-17:
@nardev do you use the latest dev/platform of espressif8266? Please run `pio update` command.

How does look your `platformio.ini`?

nardev, 2018-01-17:
I do.. i managed it somehow with build_flags = -Wl,-Tesp8266.flash.4m.ld

ivankravets, 2018-01-17:
@nardev 

1. Do you use the latest PIO Core 3.5?
2. Do you use the latest version of espressif8266 dev platform?

Please run
```
pio upgrade
pio update
```

nardev, 2018-01-18:
@ivankravets it's ok now. Works.. yes i always follow the latest pio

Thank you.