platformio/platform-espressif8266 #48: Can't uploadfs

jamarju, 2017-08-22:
Hello and thanks for the great tools!

I have some files in the `data` directory, which I want to upload as SPIFFS to an ESP8266-12 (nodemcuv2 with 4MB SPI flash). `pio run -t uploadfs` works and the resulting .spiffs file is 1028096 bytes long.

Code compiles fine:

```
$ pio run -v
[Tue Aug 22 21:45:50 2017] Processing nodemcuv2 (monitor_baud: 115200; platform: espressif8266; board: nodemcuv2; framework: esp8266-rtos-sdk)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `monitor_baud` option in `[env:nodemcuv2]` section
Collected 1 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <i2s> (/Users/javi/V/DEV/esp8266-c64-tape-emulator/firmware/lib/i2s)
xtensa-lx106-elf-size -B -d .pioenvs/nodemcuv2/firmware.elf
text	   data	    bss	    dec	    hex	filename
277214	   2126	  27352	 306692	  4ae04	.pioenvs/nodemcuv2/firmware.elf
========================================================================== [SUCCESS] Took 1.48 seconds ==========================================================================
```

But, when trying to upload I get this error:

```
$ pio run -t uploadfs
[Tue Aug 22 21:46:26 2017] Processing nodemcuv2 (monitor_baud: 115200; platform: espressif8266; board: nodemcuv2; framework: esp8266-rtos-sdk)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `monitor_baud` option in `[env:nodemcuv2]` section
Verbose mode can be enabled via `-v, --verbose` option
*** [uploadfs] IndexError `list index out of range' trying to evaluate `${SOURCES[1]}'
=========================================================================== [ERROR] Took 0.90 seconds ===========================================================================
```

jamarju, 2017-08-22:
If I use the ld script for 3M code / 1M spiffs as documented in [the docs](http://docs.platformio.org/en/latest/platforms/espressif8266.html)...

```
build_flags = -Wl,-Tesp8266.flash.4m1m.ld
```

... then I get a different set of errors when compiling:

```
$ pio run
[Tue Aug 22 21:55:14 2017] Processing nodemcuv2 (monitor_baud: 115200; platform: espressif8266; build_flags: -Wl,-Tesp8266.flash.4m1m.ld; board: nodemcuv2; framework: esp8266-rtos-sdk)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `monitor_baud` option in `[env:nodemcuv2]` section
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <i2s>
Compiling .pioenvs/nodemcuv2/src/main.o
Compiling /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/driver/driver/gpio.o
Compiling /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/driver/driver/hw_timer.o
Compiling /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/driver/driver/i2c_master.o
Compiling /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/driver/driver/spi_interface.o
Compiling /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/driver/driver/uart.o
Compiling .pioenvs/nodemcuv2/lib/i2s/i2s_freertos.o
Archiving /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libdriver.a
Archiving .pioenvs/nodemcuv2/lib/libi2s.a
Indexing /Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libdriver.a
Indexing .pioenvs/nodemcuv2/lib/libi2s.a
Linking .pioenvs/nodemcuv2/firmware.elf
/Users/javi/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/nodemcuv2/firmware.elf section `.text' will not fit in region `iram1_0_seg'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o):(.text+0x0): undefined reference to `SPI_page_program'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIWrite':
(.text+0x2f): undefined reference to `SPI_page_program'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIWrite':
(.text+0x60): undefined reference to `SPI_page_program'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIWrite':
(.text+0x91): undefined reference to `SPI_page_program'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIWrite':
(.text+0xce): undefined reference to `SPI_page_program'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIWrite':
(.text+0xdc): undefined reference to `SPI_read_data'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIWrite':
(.text+0xf0): undefined reference to `SPI_read_data'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIRead':
(.text+0x10c): undefined reference to `SPI_sector_erase'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `SPIRead':
(.text+0x13c): undefined reference to `SPI_sector_erase'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `flash_gd25q32c_write_status':
(.text+0x88c): undefined reference to `Enable_QMode'
/Users/javi/.platformio/packages/framework-esp8266-rtos-sdk/lib/libmain.a(spi_flash.o): In function `flash_gd25q32c_write_status':
(.text+0x89b): undefined reference to `Enable_QMode'
collect2: error: ld returned 1 exit status
*** [.pioenvs/nodemcuv2/firmware.elf] Error 1
=========================================================================== [ERROR] Took 2.01 seconds ===========================================================================
```

ivankravets, 2017-09-23:
Does it work with http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version ?

closed 2018-01-02 by ivankravets