platformio/platform-espressif8266 #49: Howto use dedicated stage version

to-scho, 2017-09-04:
Is there a way to install and select a dedicated commit# version from git? I want to use 2.4.0-RC1 but not floating latest git version. Regards Tobias

universam1, 2017-10-17:
same problem here, the git version bumped to rc2 which breaks a lot of stuff. So Staging is currently broken, can we somehow fix to a specific commit?

ivankravets, 2017-10-17:
@universam1 See  https://travis-ci.org/esp8266/Arduino

How to reproduce your issue?

universam1, 2017-10-17:
thanks @ivankravets here is my log output https://travis-ci.org/universam1/iSpindel

Currently ESPAsync is not compatible with 2.4 RC2, so my question is can we pin the stage to a commit like RC1 as a known good state?

universam1, 2017-11-01:
Looks like I have to move away from PIO @ivankravets as building is broken now for a longer time. There are a number of open issues with https://github.com/me-no-dev/ESPAsyncTCP being outdated but important and we can't pin to a commit with PIO. So the only solution is using Arduino again, so sad ...

ivankravets, 2017-11-01:
Could you try to use 
```
lib_deps =
  https://github.com/me-no-dev/ESPAsyncTCP.git
```

Does it work with the development version?

ivankravets, 2017-11-01:
@me-no-dev could you publish a new release of https://github.com/me-no-dev/ESPAsyncTCP ? 

Thanks!

universam1, 2017-11-01:
@ivankravets This actually untied the knot, I'm extremly happy! 😄 
Although it does not solve the original request, I can live with it for now and hope this does not return!

Thank you alot!

me-no-dev, 2017-11-01:
@ivankravets done: https://github.com/me-no-dev/ESPAsyncTCP/commit/94c7dfe1449274681680a9a1b890defde891d95c

pfabri, 2017-11-01:
I'm really glad to see that a specific problem has been solved, nonetheless I would be very keen to know if there is a solution to the broader, original question:

**Is it possible to use a specific staging version (like RC1) ?**

Many thanks in advance!

ivankravets, 2017-11-01:
@me-no-dev Thanks a lot!

@pfabri it is possible

1. Fork https://github.com/platformio/platform-espressif8266
2. Checkout to `feature/stage` branch
3. Edit [this line](https://github.com/platformio/platform-espressif8266/blob/feature/stage/platform.json#L44) and replace something with this `"version": "https://github.com/esp8266/Arduino.git#TAG"`, where TAG can be commit, branch or tag.

For example,

*  `"version": "https://github.com/esp8266/Arduino.git#e68e3402fac85af65bf28473b4fa8dd8e0341763"`

ivankravets, 2017-11-01:
I forgot, now in your `platformio.ini` use 
```
[env:...]
platform = https://github.com/MyUserName/platform-espressif8266.git#feature/stage
```

pfabri, 2017-11-02:
@ivankravets 
That's really great, (and quick!) thank you for your help.

tablatronix, 2018-01-28:
Why is this so hard? Surely there is an easier way to do this ?
Can we not just checkout any commit, and not have it overwritten by build process or something, or specify framework directory in platformio.ini

mutex36, 2018-02-17:
So I tried the following:

`platform = https://github.com/platformio/platform-espressif8266.git#feature/stage`

But it won't run. 
> Unknown development platform 'espressif8266_stage'

minida28, 2018-12-10:
For me, I need one more step:
In terminal, run `pio update`

> @me-no-dev Thanks a lot!
> 
> @pfabri it is possible
> 
> 1. Fork https://github.com/platformio/platform-espressif8266
> 2. Checkout to `feature/stage` branch
> 3. Edit [this line](https://github.com/platformio/platform-espressif8266/blob/feature/stage/platform.json#L44) and replace something with this `"version": "https://github.com/esp8266/Arduino.git#TAG"`, where TAG can be commit, branch or tag.
> 
> For example,
> 
> * `"version": "https://github.com/esp8266/Arduino.git#e68e3402fac85af65bf28473b4fa8dd8e0341763"`
> 
> Now in your `platformio.ini` use
> 
> ```
> [env:...]
> platform = https://github.com/MyUserName/platform-espressif8266.git#feature/stage
> ```