platformio/platform-espressif8266 #50: ERROR warning: can't read source directory

skymike, 2017-09-12:
I tried everything but i'm not able to upload spiffs to my esp12e, any help would be appreciated 

also reinstalled Platformio core and still not working 

`PS C:\Users\Michael\Documents\PlatformIO\Projects\170906-221237-nodemcuv2> platformio run -t uploadfs[09/13/17 01:23:43] Processing nodemcuv2 (data_dir
: C:\Users\Michael\Documents\PlatformIO\Projects\170906-221237-nodemcuv2\src\data; build_flags: -Wl,-Teagle.flash.4m1m.ld; platform: espressif8266_stag
e; upload_speed: 9600; board: nodemcuv2; framework: arduino; upload_resetmethod: ck; board_flash_mode: dio; board_f_flash: (80000000L)`
`------------------------------------------------------------------------------------------------------------------------------------------------------ Detected non-PlatformIO `data_dir` option in `[env:nodemcuv2]` section
Verbose mode can be enabled via `-v, --verbose` option
Building SPIFFS image from 'data' directory to .pioenvs\nodemcuv2\spiffs.bin
warning: can't read source directory
*** [.pioenvs\nodemcuv2\spiffs.bin] Error 1
============================================================= [ERROR] Took 0.68 seconds =============================================================`

ivankravets, 2017-09-23:
Do you have `data` directory in project? Does it contain any files?

ivankravets, 2018-01-02:
Please reopen if you still need help

closed 2018-01-02 by ivankravets

spilz87, 2024-05-13:
I had the same issue
I solved it by adding an empty folder in my project

spilz87, 2024-05-13:
I had the same issue
I solved it by adding an empty folder « data » in my project