platformio/platform-espressif8266 #51: mkspiffs outdated - breaks staging!

universam1, 2017-09-15:
Staging updated SPIFFS to 0.3.7 which is not compatible with the tool mkspiffs that is currently deployed

tool-mkspiffs                   @ 1.102.0        [Up-to-date]

As a result now there is a terrible condition that mkspiffs is breaking the SPIFFS from staging rendering it unusable.
Please update the tool-mkspiffs.

ivankravets, 2017-09-15:
Could you run `pio update` and try to build a project again?

universam1, 2017-09-15:
Updating espressif8266_stage             @ d192958        [23e7335]
Updating framework-arduinoespressif8266  @ e68e340        [3e9caf7]
PackageManager: Installing tool-mkspiffs @ ~1.103.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-mkspiffs @ 1.103.0 has been successfully installed!

Thanks for quick response! 👍 

Unfortunately, it turns out to be an incompatibility issue with older version: https://github.com/igrr/mkspiffs/issues/26
Lets see how this will be solved and if means to require different versions btw. staging and normal version.

ivankravets, 2017-09-15:
@universam1 should we do something on our side?

universam1, 2017-09-15:
Well, I think we should put it on hold until the Arduino team decides how they will solve this conflicts. Looks like we have no other solution.

universam1, 2017-09-22:
suggesting updating the tool once again https://github.com/esp8266/Arduino/commit/1ee17e54dd1fb58fe74456c3e6c3c3256dd047a5 and with https://github.com/esp8266/Arduino/commit/1683b12dd7105f456530931709e69f12e2994d6e should be fixed

Lan-Hekary, 2017-09-22:
I was just going in in post that same comment @universam1  :D

ivankravets, 2017-09-23:
Done! Please `pio update`

closed 2017-09-23 by ivankravets

ivankravets, 2017-09-25:
@universam1 Thanks! Fixed.

universam1, 2017-09-26:
nice work @ivankravets lot of things fixed at once! 👍