platformio/platform-espressif8266 #54: ENV PROGNAME can't be overriden

universam1, 2017-10-02:
I was redirected here from platformio/platformio-core/issues/1090

What kind of issue is this?

- [x ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
win10 64bit

**PlatformIO Version** (`platformio --version`):
3.5.0a16

### Description of problem
Followed guidance from https://community.platformio.org/t/rename-bin-file-on-build/2390

Nevertheless, the file name does not change from "firmware.bin"

#### Steps to Reproduce

1. add platform.ini `extra_scripts = extra_script.py`
2. edit extra_script.py
```
Import("env")
build_tag = "v2.2"
env.Replace(PROGNAME="test%s" % build_tag)
print env['PROGNAME'] 
```
3. Prints out `testv2.2`

### Actual Results
Prints out `testv2.2`
but firmware filename stays "firmware.bin"

```
testv22
Linking .pioenvs\nodemcuv2\firmware.elf
Calculating size .pioenvs\nodemcuv2\firmware.elf
Building .pioenvs\nodemcuv2\firmware.bin
```
### Expected Results
filename should be "testv2.2.bin"

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
extra_scripts = extra_script.py
```

**The content of `extra_script.py`:**
```py
Import("env")

build_tag = "v22"

env.Replace(PROGNAME="test%s" % build_tag)
print env['PROGNAME'] 
```

universam1, 2017-10-02:
Maybe it's being hardcoded here https://github.com/platformio/platform-espressif8266/blob/921855a9c530082efddb5d48b44c3f4be0e2dfa2/builder/main.py#L144

ivankravets, 2018-01-03:
Please note that it should be PRE script
```
extra_scripts = pre:extra_script.py
```

commit 9f345e3f (2018-01-03): Allow to override PROGNAME // Resolve #54
closed 2018-01-03 by ivankravets