platformio/platform-espressif8266 #58: Stack Trace Decoding

cyberjacob, 2017-11-22:
From the Forums, at https://community.platformio.org/t/could-we-add-support-for-stack-trace-decoding-in-the-esp8266/1017

The Arduino IDE supports decoding the stack traces that the ESP8266 outputs through the serial connection https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md#Decode79. Would it be possible to integrate this tool into PlatformIO: https://github.com/me-no-dev/EspExceptionDecoder40.

Loufylouf, 2017-11-27:
I've tested the command line interface of the tool mentioned by @cyberjacob , found at https://github.com/littleyoda/EspStackTraceDecoder and this works pretty great. It doesn't seem that hard to integrate as it only requires : 

- Java (openjdk-9-jre works for me)
- xtea-lx106-elf-addr2line (located at ~/.platformio/packages/toolchain-xtensa/bin/ for me)
- firmware.elf
- file containing the dump of the stack trace

I bet the first three are pretty easy to get, the only remainder is the dump (pio device monitor --port /dev/ttyUSB0 > dump.txt did not work because of the garbage sent on Serial when doing a reset of the board, so I copy-pasted it).

cyberjacob, 2017-11-27:
The original Arduino IDE plugin solves this by requiring the user to copy/paste the trace into it. However, it works with extra garbage data at the start/end, which implies there's a regex or similar involved to find the actual data. If we could find this, could platformio scan the output for a match and go from there?

cyberjacob, 2017-11-27:
Found the regexes: [me-no-dev/EspExceptionDecoder/src/EspExceptionDecoder.java#L349](https://github.com/me-no-dev/EspExceptionDecoder/blob/master/src/EspExceptionDecoder.java#L349) and [me-no-dev/EspExceptionDecoder/src/EspExceptionDecoder.java#L362](https://github.com/me-no-dev/EspExceptionDecoder/blob/master/src/EspExceptionDecoder.java#L362)

Loufylouf, 2017-11-28:
The issue was already documented and there's a bounty on it https://github.com/platformio/platform-espressif8266/issues/31 if you're interested @cyberjacob .

ivankravets, 2018-01-03:
Duplicate of https://github.com/platformio/platform-espressif8266/issues/31

closed 2018-01-03 by ivankravets