platformio/platform-espressif8266 #61: stage is broken: Error: Detected unknown package 'espressif8266_stage'

simap, 2017-11-30:
The instructions here:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

don't work anymore. Commit 
https://github.com/platformio/platform-espressif8266/commit/a05b8108a606fd1035333619d798ed2362ba9d52
was merged into feature/stage and took out the "_stage" part. trying to build now gives:
```
Error: Detected unknown package 'espressif8266_stage'
```

middelink, 2017-12-02:
It looks like they want us to use the new syntax...
https://github.com/platformio/platform-espressif8266.git#feature/stage

simap, 2017-12-23:
@middelink that does't seem to work for me and results with this error:
```
Error: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage'

```

ivankravets, 2018-01-02:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

Please check that you use PIO Core 3.5

closed 2018-01-02 by ivankravets