platformio/platform-espressif8266 #62: esp8266 rtos sdk example is broken

vshymanskyy, 2017-12-03:
This example builds OK, but fails to run on ESP8266 (NodeMCU):
https://github.com/platformio/platform-espressif8266/tree/develop/examples/esp8266-rtos-sdk-blink
When uploaded to the board, it just prints lots of garbage...

ivankravets, 2017-12-04:
@yanbe could you take a look at this issue?

Djowietje, 2018-01-09:
You have a while(true) loop, doing so will crash cause the watchdog timer to kick in and reset your board.

umitech, 2018-10-03:
The same problem here. But When I changed the baudrate to 76800, I could see that the board is continually resetting and print this:

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 31360, room 16
tail 0
chksum 0xcb
load 0x3ffe8000, len 2128, room 8
tail 8
chksum 0x7e
load 0x3ffe8850, len 480, room 0
tail 0
chksum 0xaf
csum 0xaf
OS SDK ver: 1.5.0-dev(caff253) compiled @ Oct 23 2017 17:42:20
rf_cal[0] !=0x05,is 0xFF

RTOS is a great feature. It would be great if you can solve it.

mbmcavoy, 2018-12-23:
I'm new to using ESP8266 (and PlatformIO) and wanting to use the RTOS SDK as well.

I believe the problem is that the default firmware on the NodeMCU boards is not compatible with the ESP8266 SDK. However, my initial experiments were in using the Arduino IDE, and using the Arduino Framework within PlatformIO also worked. It's possible flashing with the Arduino tools overwrote the needed files.

I was able to get the RTOS SDK example working by installing esptool, and manually flashing the files from the SDK folder to the correct locations:

The files are in `C:\Users\$user\.platformio\packages\framework-esp8266-rtos-sdk\bin`.
`> esptool.py erase_flash`
`> esptool.py write_flash 0x3fc000 esp_init_data_default.bin`
`> esptool.py write_flash 0x3fe000 blank.bin`

Initially, I also manually flashed the compiled files in my project folder under `.pioenvs\nodemcuv2\`.
`> esptool.py write_flash 0x00000 eagle.flash.bin`
`> esptool.py write_flash 0x20000 eagle.irom0text.bin`

At this point, the program started working (the LED is flashing). I can also modify the example, compile, and upload from within PlatformIO successfully!

Oddly, I have not been able to get the Non-OS SDK example working, even by flashing the files from that SDK, but since I want the RTOS, I haven't dug too far. I suspect I just made an error trying it...

zenz, 2018-12-24:
@mbmcavoy  Thank you very much! your comment is very useful!

randomhammer, 2019-03-03:
I am having the same problem. I was able to get the blink example to compile and upload by manually flashing the firmware, but I still get rubbish from the serial monitor. Keep trying different speeds but nothing seems to work. Any well to tell what speed the ESP8266 is expecting to talk at?

maxgerhardt, 2019-03-05:
Directly related to https://github.com/platformio/platform-espressif8266/issues/133

ivankravets, 2019-03-06:
Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions

Does it work now?

ivankravets, 2019-03-06:
Thanks @mbmcavoy for the research. It has been fixed in the latest build.

commit ab839a29 (2019-03-06): Fix issue with broken examples for NONOS SDK and RTOS SDK // Resolve #62 Resolve #133
commit ce5b99b4 (2019-03-06): Flash blank.bin automatically for NONOS SDK and RTOS SDK // Resolve #62 Resolve #133
closed 2019-03-06 by ivankravets