platformio/platform-espressif8266 #63: Failed build with platform=https://github.com/platformio/platform-espressif8266.git#feature/stage

PhracturedBlue, 2017-12-03:
I recently updated my platformio and now
```
platform=espressif8266_stage
```
 has been replaced with
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
```
But this doesn't work for me either.

I created a new virtualenv, ran `pip install platformio`, erased ~/.platformio and .pio*/ and then ran
`pio run --verbose`

The results look like this:
```
[Sun Dec  3 15:43:51 2017] Processing nossl (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; lib_deps: ESP8266MQTTMesh; build_flags: -DLED_PIN=2 -g; board: esp01_1m; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: /home/piotest/src/ESPtest/.piolibdeps
LibraryManager: Installing id=1747
ESP8266MQTTMesh @ 0.8.7 is already installed
Installing dependencies
Looking for ESPAsyncTCP library in registry
Found: http://platformio.org/lib/show/305/ESPAsyncTCP
LibraryManager: Installing id=305 @ ^1.1.3
ESPAsyncTCP @ 1.1.3 is already installed
Looking for AsyncMqttClient library in registry
Found: http://platformio.org/lib/show/346/AsyncMqttClient
LibraryManager: Installing id=346 @ ^0.8.1
AsyncMqttClient @ 0.8.1 is already installed
Installing dependencies
Looking for ESPAsyncTCP library in registry
Found: http://platformio.org/lib/show/305/ESPAsyncTCP
LibraryManager: Installing id=305 @ ^1.0.1
ESPAsyncTCP @ 1.1.3 is already installed
PlatformManager: Installing platform-espressif8266
espressif8266 @ 4f59cef is already installed
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
Error: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage'
```

platformio.ini:
```
[platformio]
env_default = nossl

[env:nossl]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
#platform = espressif8266
#platform = espressif8266@4f59cef
board = esp01_1m
framework = arduino
lib_deps = ESP8266MQTTMesh
build_flags = -DLED_PIN=2 -g
```
I am able to build if I use `platform = espressif8266` or if I use `platform = espressif8266@4f59cef` but not with `platform = https://github.com/platformio/platform-espressif8266.git#feature/stage`

ivankravets, 2018-01-02:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

Or switch to stable 2.4.0.

closed 2018-01-02 by ivankravets

PhracturedBlue, 2018-01-02:
I don't understand how that link is different than what I did.  I followed the instructions in that link prior to submitting the ticket.

Now that 2.4.0 is released, I can indeed use that without needing the git link, but it doesn't really answer the question in the ticket of how I can get the git link to work properly.

ivankravets, 2018-01-02:
1. It seems that you don't use the latest stable PIO Core 3.5. Please `pio upgrade`

2. We know about 2.4.0 and working on the new release for dev/platform