platformio/platform-espressif8266 #69: Unable to build SPIFFS for ESP8266 d1_pro_mini

VonLatvala, 2018-01-18:
What kind of issue is this?

Unable to build SPIFFS for ESP8266 d1_pro_mini

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS High Sierra 10.13.2 (17C88)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.0

### Description of problem

The process of building the SPI FFS fails. The error message is as follows:

```
libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector
*** [.pioenvs/d1_mini_pro/spiffs.bin] Error -6
```

#### Steps to Reproduce

1. `git clone git@bitbucket.org:VonLatvala/esp8266_platformio_spiffs_problem.git`
2. `pio run --target buildfs`

### Actual Results

```
libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector
*** [.pioenvs/d1_mini_pro/spiffs.bin] Error -6
```

No FS image was built.

### Expected Results

FS image built successfully.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
```

**Source file to reproduce issue:**
I have a blank main.cpp with setup&loop in the git repository. Should not matter.

### Additional info

VonLatvala, 2018-01-19:
Wonderful, thank you! I am running the develop version of platformio (PlatformIO, version 3.5.2a1), and tried updating the platform espressif8266 to incorporate this bugfix, but it tells me that the platform is up to date:
```
platformio platform update espressif8266
Updating espressif8266                   @ 1.6.0          [Up-to-date]
```

How do I force the platform to use develop branch? I'd really like to use this feature ASAP.

ivankravets, 2018-01-19:
Take a look at the README of this repo

VonLatvala, 2018-01-19:
Thanks, that worked!

Jaykob, 2024-04-09:
Doesn't seem to be fixed until today. I'm getting the same error in the current stable version...