platformio/platform-espressif8266 #73: Interrupt trigger or digitalRead resets ESP8266 after deepSleep()

gmdriscoll, 2018-02-17:
###Basic Infos

After compiling and uploading a sketch, a digitalRead of a pin or an interrupt on a pin will do what it is supposed to. After a deepSleep(), the same event will reset the ESP8266. This example was done with the interrupt routine as that is the desired use. A simple loop polling the pin with digitalRead() has the same effect after returning from sleep.

Produces the same effect when using Arduino IDE.

###Compiler 1

Using PlatformIO on VSCode with latest updates as far as I can tell.
VS Code 1.20.1
Shell 1.7.9
Renderer 58.0.3029.110
node 7.9.0
Architecture x64
Settings in Compiler 1 - PlatformIO IDE

Processing nodemcuv2 (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: nodemcuv2; framework: arduino)
PLATFORM: Espressif 8266 (Stage) > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 25 compatible libraries
Scanning dependencies...
Dependency Graph
|-- v1.0
Linking .pioenvs\nodemcuv2\firmware.elf
Calculating size .pioenvs\nodemcuv2\firmware.elf
text data bss dec hex filename
255878 3392 30104 289374 46a5e .pioenvs\nodemcuv2\firmware.elf

###Compiler 2

Arduinio 1.8.5
NodeMCU ESP8266 1.0 ESP-12E
Flash Size: 4M(3M SPIFFS)
Debug Port: Disabled
Debug Level: None
IwIP Variant: v2 prebuilt (MSS=536)
CPU: 80mhz
Upload Speed: 921600
http://arduino.esp8266.com/stable/package_esp8266com_index.json

###Hardware
Hardware: NodeMCU ESP8266 1.0 ESP-12E
Core Version: ??? Not sure in PlatformIO but setup per instructions to get directly from github

###Wiring

REQUIRED - Pin D0(GPIO 16) tied to reset
TRIED - On board Flash button with D3(GPIO 0) disconnected and held high with 1k resistor
TRIED - D5(GPIO 14) with external switch to ground, held high with internal pullup and external 1k pullup
TRIED - D1(GPIO 5) with external switch to ground, held high with internal pullup and external 1k pullup
TRIED - D3 (GPIO 0) and D4(GPIO 2) tied high and D8(GPIO 15) tied low

###Sketch

#include <Arduino.h>
#include <ESP8266WiFi.h>

#define FLASHPIN 14 // GPIO 14 , Pin D5 On NodeMCU ESP8266-12E
//#define FLASHPIN 0

// =================== DEBUG PRINT DEFINITIONS ======================
#define DEBUG1 // comment out to remove print statements
#if defined(DEBUG1)
#define debug_print(a) (Serial.print(a))
#define debug_println(a) (Serial.println(a))
#else
#define debug_print(a)
#define debug_println(a)
#endif

unsigned int deepSleep = 20;

void CheckResetFlashButtonInterrupt(){
debug_println("Interrupt Button Pressed...");
digitalWrite(LED_BUILTIN, LOW);
debug_println("LED SET...Waiting 2 sec...");
// yield(); // Causes WTD RESET Cause 4
// delay(2000); // 2sec
debug_println("Writing RESET BIT to eeprom...");
digitalWrite(LED_BUILTIN, HIGH);
}

void setup() {
Serial.begin(115200);
delay(500); // added to stabilize initial serial output

pinMode(LED_BUILTIN, OUTPUT); // Initialize the LED_BUILTIN pin as an output
digitalWrite(LED_BUILTIN, HIGH);
pinMode(FLASHPIN, INPUT_PULLUP); //set up the Flash pin

attachInterrupt(digitalPinToInterrupt(FLASHPIN), CheckResetFlashButtonInterrupt, CHANGE);

// wait for 15 sec for button to be pressed
int i=0;
for( i = 0; i < 16; i++ ){
debug_println("Waiting for interrupt to be pressed");
yield();
delay(1000); // 1 sec
}

ESP.deepSleep(deepSleep * 1000000); //Sleep ESP8266 for DEEPSLEEP TIME uSec
}

void loop() {
// put your main code here, to run repeatedly:

}

###Debug Description

-The interrupt routine was entered right after upload before the 1st sleep.
-The first sleep reset was the first bit of .
-Subsequent is from pressing the button and the system resetting instead of entering the interrupt routine.

###Debug Messages

Waiting for interrupt to be pressed
Interrupt Button Pressed...
LED SET...Waiting 2 sec...
Writing RESET BIT to eeprom...
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
{$␀$��|␀�l�|␂␌␌␌�␄$�␌"|��␃�␒�r�#�␄#��No�lnN���␌b␜p��#l␂rlp�n�␐␂␌␌�␄d␌��␄␄␌b␄o�|␃l�␄␌�p␄��nn�␀$��␀l ␃�␛␛No␄l␂␎␂or�ےo␄␌#␄�␎$␏{��o␌␌b␄�␏$�␂␄␌␌␌␄␄␄l␂��o�␂Waiting for interrupt to be pressed Waiting for interrupt to be pressed Waiting for interrupt to be pressed Waiting for interrupt to be pressed Irl␀l��|␀�c␃�N␌␌␌�␄l�␌B|��␂�␒�{��>�␄b��NN�␀loN���␌c␜p��cl␂rlp�o�␐␂␌␌��$␌��␄␄␌b␌o�|␂l�␌␄�b�|␡�N�␃l��$␃�␛␛oo␄l␃␏␂Nr���o␄␄c␌�␏l␎s��N␄␌c␌�␏l�␃␌␌␄␄␌␌␌d ␃��o�␂I�{l␀lܞ�␂�l�|␃␄␄␄�␜�l�␄b|��␂�␒�{��>�␄b�p␡�N�loo���␄b␜p��$rlsdp�N�␐␃␌␌�␄$␌��␌␄␄"␌o�|␂$�␜�␌�p␄��nn�␃l��$␃�␛2oN␌l␂␎␂Ns���o␄␄c␌�␏$␎s��o␄␄c␌�␏$�␂␌␌␌␌␄␄␄l`␂��N�␂Waiting for interrupt to be pressed
Waiting for interrupt to be pressed
Waiting for interrupt to be pressed

ivankravets, 2018-02-24:
Do you mean that you can reproduce this issue with Arduino IDE too?

gmdriscoll, 2018-02-25:
To answer your question specifically, Yes, it happens on both IDE's.  The problem is not what I thought, though.  The problem is that the builtin LED is tied/multiplexed to D0 (GPIO 16).  When using deepSleep(), the ESP8266 drives this pin low to reset when awakening from deep sleep.   By making this pin an output and driving it low with the interrupt routine to turn on the LED, the same pin used to reset the Esp8266 after wakeup is driven low directly and causes the reset.

It didn't occur to me for quite a while that D0 was used as the LED pin.   If there is another way to use the builtin LED so it didn't cause a reset, please let me know, but at this point it seems to be a hardware use issue.

ivankravets, 2018-04-20:
> To answer your question specifically, Yes, it happens on both IDE's.

Please file an issue here https://github.com/esp8266/Arduino/issues

closed 2018-04-20 by ivankravets