platformio/platform-espressif8266 #74: Settings for ESP8266MOD ESP-12F QIO L4

dontsovcmc, 2018-02-24:
What should I choose for raw ESP8266MOD Ai-thinker ESP-12F QIO module?
Chip ID: 0x00167881
flash_id:
Manufacturer: e0
Device: 4016

Should I **erase_flash** by esptool for the first time?

Thanks a lot!

bevchou, 2018-04-19:
I have the same question - any one found a solution?

ivankravets, 2018-04-20:
Could you program this board with Arduino IDE?

bevchou, 2018-04-20:
I was able to get code to upload (just the blink sketch) using the Arduino IDE, but it is not able to run any of the code since it is stuck in a reset loop. I am not sure if this is caused by improper settings or if it is a hardware problem.

rescalda, 2018-04-29:
I lost several nights with the same problem on ESP-12-F QIO L4 and, finally, I discovered that it **does not work** in QIO mode but only in **DIO** mode! I have 4 of them, same batch from Aliexpress and all behave in the same way. In addition, I discovered that, should you read ADC, either via analogInput(A0) or ESP.getVcc() the radio section fails and WiFi is no longer available!

bigbuglamer, 2018-06-29:
Radio section fails because you are reading too intensive without yield() in between. And yes, there is a tonns of esp-12-f marked as qio but dio actually. So ... just use dio mode.

rescalda, 2018-06-29:
> Radio section fails because you are reading too intensive without 
No, it is enough one read and the radio fails forever. At lest, this is what happens on on 4 ESP12-F

>here is a tonns of esp-12-f marked as qio but dio actually
Of course we must use DIO or discard them. It is a matter of fact that it is a fraud to sell them as QIO. 
In addition, if they are DIO, I should have a couple of extra GPIO pins available for me, but this looks not true too.

bigbuglamer, 2018-08-09:
Now there is a service to check modules: https://www.ai-thinker.com/service/autifake