platformio/platform-espressif8266 #75: Please add support to 4m2m LD Script

capedra, 2018-03-02:
@ivankravets 
Hello, could you please add support to 4m2m LD script for ESP boards? Could you please implement that for us?

According to the following links, it's possible to do that:
https://github.com/esp8266/Arduino/issues/3226
http://www.esp8266.com/viewtopic.php?f=32&t=14639

Thanks!!

ivankravets, 2018-03-02:
Please make PR to https://github.com/esp8266/Arduino as @igrr requested here https://github.com/esp8266/Arduino/issues/3226#issuecomment-300469745

capedra, 2018-03-03:
@ivankravets @igrr
I don't know how to do it. It's too complicated. 😐
Could you please help me out and open it? I think it will be helpful for people who wants to use OTA and SPIFFS at the same time.
Best Regards!

bigbuglamer, 2018-06-29:
it's working for me with recent versions. Maybe issue can be closed?

ivankravets, 2018-06-29:
Thansk!

closed 2018-06-29 by ivankravets