platformio/platform-espressif8266 #76: When will ESP core for Arduino 2.4.1 be released here?

arendst, 2018-03-12:
Any plans for releasing 2.4.1 in the near future?

bigbuglamer, 2018-03-12:
Yeah, please.

tablatronix, 2018-03-14:
ugh serisouly

reloxx13, 2018-03-28:
@arendst 
i used 
`platform = https://github.com/platformio/platform-espressif8266.git#feature/stage ;latest git version of esp8266 arduino`

to get the latest version, but could not get it to work with tasmota for now :X

jackjansen, 2018-03-28:
@reloxx13 does that give you the upstream 2.4.1? I get the impression it does not (from the commit messages)... (I want 2.4.1 myself because I need the https web server).

reloxx13, 2018-03-28:
@jackjansen yes and no, it takes the latest git version from expressif.
sometimes i had to clear the libs myself in the %appdata% .platformio folder and build again.

also, dont forget to update your libs from the top platformio menü entry.

tablatronix, 2018-03-28:
I have been installing staging and manually pulling from git, seems to work well.

tablatronix, 2018-03-29:
Yeah I manually git checkout the 2.4.1 tag, I think it works

ivankravets, 2018-04-20:
Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

closed 2018-04-20 by ivankravets

arendst, 2018-04-23:
Thanks for releasing 1.7.0 today. It at least seems to compile with ESP/Arduino 2.4.1 but...

- It seems to default to displaying all warnings which I'm unable to turn off
- It disregards my platformio.ini loader file without a SPIFFS area and insists to use a SPIFFS of 256k
```
build_flags = -Wl,-Tesp8266.flash.1m0.ld
```

This makes it unusable to me.

ivankravets, 2018-04-23:
1. http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-unflags -> `build_unflags = -Wall` or use http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags and disable what do you need
2. http://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

mcspr, 2018-04-23:
@ivankravets But it did use 1m0 with 1.6.0, somehow substituting esp8266 for eagle.
Tried out with travis and dummy esp01_1m project: https://travis-ci.org/mcspr/test-ldscript/builds/370240022
1.5.0 - last line has `-Wl,-Tesp8266.flash.1m0.ld` (i assume from project root)
1.6.0 - `-Wl,-Teagle.flash.1m0.ld` (how?)
But with 1.7.0 it just uses board default (via esp01_1m.json -> 1m256)

arendst, 2018-04-25:
Still unable to change the default from1M256 to 1M0. It works fine in 1.6.0. It doesn't work in 1.7.0 as @mcspr also noted.

- Tried reverting changes between 1.6.0 and 1.7.0 in ``platform-build.py`` with no success.
- Found test for user provided flash.ld in ``builder/main.py`` and copied 1.6.0 ``main.py`` to 1.7.0 ``main.py`` with still no luck.

There must be a change somewhere preventing me as a user to change the default 1M256 to 1M0. Very frustrating as it prevents releasing a new version of Tasmota (as I do not want users to complain about unability to OTA upgrade anymore caused by 1.7.0 - so out of my hands).

ivankravets, 2018-04-25:
I’ll take a look at this today. Sorry.

ivankravets, 2018-04-25:
@arendst sorry for the issue. That was a bug and I fixed it. Please `pio update`. New users will not be affected.

ivankravets, 2018-04-25:
P.S: You can also use old format `build_flags = -Wl,-Tesp8266.flash.1m0.ld`

arendst, 2018-04-25:
Thank you very much for the quick reply and fix.

Just tested it and It now works as expected.

commit ce9bc39a (2018-04-20): Update Arduino Core to 2.4.1 // Resolve #76
commit 5882d3a3 (2018-04-20): Merge branch 'develop' into feature/stage

* develop:
  Update Arduino Core to 2.4.1 // Resolve #76
  Don't clone build environment

# Conflicts:
#	platform.json