platformio/platform-espressif8266 #77: WiFi broken since v1.6.0

chkr1011, 2018-03-14:
Hi,
I am using this platform with some ESP12E/F based devices like NodeMcu and since the release of version 1.6.0 the WiFi module is no longer working properly.

With version 1.5.0 the code runs fine since lots of month. The device is connecting and stays connected. After upgrading to version 1.6.0 the WiFi still connects but every 5-10 seconds the connection is lost and after additional 5-10 seconds it is connected again. The device is not crashing and I am counting the reconnects.

My platformio.ini looks like this:
```ini
[env:default]
platform = espressif8266 # Works when using espressif8266@1.5.0
board = esp12e
framework = arduino
upload_speed = 115200
build_flags = -DMQTT_MAX_PACKET_SIZE=512
lib_deps = ArduinoJSON, PubSubClient, RC-Switch
```

The code which is responsible for maintaining the WiFi connection looks like this:
```c
#include "wifi-settings.h"

String WIFI_HOSTNAME = "";

uint8_t _status = WL_DISCONNECTED;
uint16_t _wifiReconnectCount = 0;

bool wifi_isConnected() { return WiFi.status() == WL_CONNECTED; }

int wifi_getRssi() { return WiFi.RSSI(); }

uint16_t wifi_getReconnectCount() { return _wifiReconnectCount; }

void wifi_setup() {
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  WiFi.setOutputPower(0);
  ESP.eraseConfig();

#ifdef DEBUG_WH
  Serial.printf("WiFi: Connecting to '%s' (%s)...\n", WIFI_SSID, WIFI_PASSWORD);
#endif

  WiFi.mode(WIFI_STA);
  WiFi.hostname(WIFI_HOSTNAME);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void wifi_loop() {
  uint8_t status = WiFi.status();
  if (status == _status) {
    return;
  }

  _status = status;

  if (_status == WL_CONNECTED) {
    _wifiReconnectCount++;
  }

#ifdef DEBUG_WH
  Serial.printf("WiFi: Connected=%s (IP=%s)\n", _status == WL_CONNECTED ? "yes" : "no", WiFi.localIP().toString().c_str());
#endif
}
```

I also tried to erase the whole flash using the _esptool_ but it doesn't help. I am clueless because the same code without any changes works when using 1.5.0 of the platform. I also tested a couple of brand new boards. All have the same problem.

I exposed several system information using the HTTP server. Maybe this information is relevant.
```json
{
    "vcc": 65535,
    "freeHeap": 40448,
    "sdkVersion": "2.1.0(deb1901)",
    "coreVersion": "2_4_0",
    "bootVersion": 31,
    "bootMode": 1,
    "wifi_rssi": -66,
    "wifi_reconnectCount": 7,
    "mqtt_isConnected": false,
    "mqtt_reconnectCount": 3,
    "cpuFreqMHz": 80,
    "flashChipId": 1458376,
    "flashChipRealSize": 4194304,
    "flashChipSize": 4194304,
    "flashChipSpeed": 40000000,
    "flashChipMode": true,
    "flashChipSizeByChipId": 4194304,
    "sketch_size": 310912,
    "sketch_hash": "c4ec907a1b1092662eb89d163f192583",
    "sketch_freeSpace": 2834432,
    "reset_reason": "External System",
    "reset_info": "Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"
}
```

Best regards
Christian

mamama1, 2018-04-02:
similar issue here. however when i turn off serial output, wifi starts to work again. same code compiled in arduino ide (most recent one, most recent esp8266 core) makes no issues at all, even with serial output enabled.

maybe some compilation/linking order issue?

riccardo1991, 2018-04-16:
I have the same issue: the WiFi is connecting successfully if I flash with ArduinoIDE, while it remains stuck if I flash with PlatformIO. The code is the same. It was working with PlatformIO for sure until 20th of February (all updated till that date)

The platformio.ini
```
[env:d1_mini]
platform=espressif8266
board=d1_mini
framework=arduino

[platformio]
src_dir=NTPtoRTC_setter
```

ivankravets, 2018-04-20:
I've just tested and can't reproduce with the latest upstream/dev version.

ivankravets, 2018-06-04:
Please reopen if you still need help

closed 2018-06-04 by ivankravets

mamama1, 2018-06-04:
i will have time for further testing within the next weeks. I'll then also try to make a minimalistic code which triggers the issue.