platformio/platform-espressif8266 #78: Compilation error

leobel96, 2018-03-20:
I'm trying to compile the "NMEAsimple.ino" example from https://github.com/SlashDevin/NeoGPS with platformio in VSCode for NodeMCU board but I'm getting a lot of:
error: 'constexpr' needed for in-class initialization of static data member 'const float
Compiling it with Arduino IDE returns no error so, I think, it's a PlatformIO compiler error.

ivankravets, 2018-03-20:
@valeros do you have any ideas?

leobel96, 2018-03-20:
Maybe this one can help: https://stackoverflow.com/questions/9141950/initializing-const-member-within-class-declaration-in-c

leobel96, 2018-03-27:
@ivankravets no solution yet?

valeros, 2018-03-27:
Hi @leobel96 !
As a workaround, could you please try to add the next line to your `platformio.ini`
```
build_flags = -D ARDUINO=10803
```

leobel96, 2018-03-28:
@valeros Thank you!! Now it compiles!! What is this? Should I always include that line in platformio.ini file?

ivankravets, 2018-04-20:
Is fixed in upcoming release

closed 2018-04-20 by ivankravets