platformio/platform-espressif8266 #82: Including ESP8266HTTPClient pulls in ESP8266MQTTMesh and fails by not finding Ticker.h

phi1010, 2018-04-07:
The dependency resolution for the ESP8266HTTPClient library seems to fail, according to the build output it requires the MQTT Mesh library; however I can't find any reference to that library in the source code at https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPClient which would explain why this happens:

```
-- <ESP8266HTTPClient> v1.1
|   |-- <ESP8266MQTTMesh> v0.8.8
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |   |   |-- <Debug>
|   |   |   |-- <Debug>
|   |   |-- <AsyncMqttClient> v0.8.2
|   |   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |   |-- <ESP8266WiFi> v1.0
|   |   |   |   |   |-- <Debug>
|   |   |   |   |-- <Debug>
|   |   |-- <ESP8266WiFi> v1.0
|   |   |   |-- <Debug>
|   |-- <ESP8266WiFi> v1.0
|   |   |-- <Debug>
```

This also causes the compilation to fail, since the Ticker library included from the MQTT lib is not resolved, although it should be when the MQTT library is included:

```
In file included from C:\Users\USERNAME\.platformio\lib\ESP8266MQTTMesh_ID1747\src\ESP8266MQTTMesh.cpp:18:0:
C:\Users\USERNAME\.platformio\lib\ESP8266MQTTMesh_ID1747\src\ESP8266MQTTMesh.h:21:20: fatal error: Ticker.h: No such file or directory

****************************************************************
* Looking for Ticker.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Ticker.h"
* Web  > https://platformio.org/lib/search?query=header:Ticker.h
*
****************************************************************

#include <Ticker.h>
^
compilation terminated.
Compiling .pioenvs\d1_mini\FrameworkArduino\Esp.cpp.o
*** [.pioenvs\d1_mini\lib311\ESP8266MQTTMesh_ID1747\ESP8266MQTTMesh.cpp.o] Error 1
============================================================================================== [ERROR] Took 9.28 seconds ==============================================================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

I'm using the up-to-date Espressif 8266 1.6.0 platform and this configuration: 

```
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
monitor_baud = 74880
```

ivankravets, 2018-04-20:
If you don't need `ESP8266MQTTMesh`, just add this line

```ini
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
monitor_baud = 74880
lib_ignore = ESP8266MQTTMesh
```

ivankravets, 2018-04-20:
http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

phi1010, 2018-04-23:
I already did -- still, this resolution result doesn't seem sensible; and even if it is sensible to include ESP8266MQTTMesh, the Ticker library should be included too.

ivankravets, 2018-04-23:
Could you provide a simple project to reproduce this issue?

closed 2019-02-02 by ivankravets