platformio/platform-espressif8266 #83: Arduino IDE compiling differently that PlatformIO, causing bugs in ESP8266

mamama1, 2018-04-13:
### Configuration

**Operating system**: Windows 10 64 Bit

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.3a8

### Description of problem

I've got a pretty big (around 4k lines) software written for the ESP8266 (WiFi LED dimmer) which seems to compile differently between Arduino IDE and PIO.
The firmware has two TCP servers. One is a webserver and based on the ESP8266WebServer.h and the second one is a simple tcp server which listens to incoming bytes and is based on the ESP8266WiFi.h (class WiFiServer and WiFiClient).

When compiling the software using the Arduino IDE, everything works as expected. When compiling the same software with PIO, the tcp server which is based on ESP8266WiFi.h stops working after serving one request. The webserver and the rest of the devices continues working. However, also Wifi connection issues occur, like SoftAP not working (SSID doesn't show up).

I have reproduced this many times, it occurs using any ESP8266 core version beginning with 2.2.0 when I use PIO and does NOT occur on any Arduino compiled ESP8266 core, also beginning with 2.2.0 including 2.4.0-rc2.

I can't isolate the issue so I guess it will be hard to track this one down using the code. Rather, maybe it would be possible to somehow match the compile options of the Arduino IDE and PIO? Something stuff-breaking must be different here?!

I "converted" the Arduino .ino project this way:
- copy .ino content to main.cpp
- change order of functions so that no functions are called which aren't declared yet - arduino ide doesn't seem to care about the order
- put setup and loop to the end of the file

**The content of `platformio.ini`:**
```ini
[env:nodemcuv2]
platform = espressif8266@1.6.0
board = nodemcuv2
framework = arduino
monitor_baud = 115200
```

There is an addin for Visual Studio, called "VisualMicro" which had a very(!) similar or even exact the same issue. I contacted the developer and after sending him the build logs and a few months he said that there was something wrong with the linking order or something and now it seems to be fixed.
So I don't think that this is an error on my side, although I wouldn't be opposed to that if it were so.

Please help, i think PIO is REALLY cool stuff, but this is really stuff breaking and I can't even use it as an IDE and all the cool lib manager/dependency features because the Arduino IDE can't compile PIO stuff.

Thanks!

mamama1, 2018-04-13:
This is the Arduino IDE buildlog with all the linking and build options. This compiled sketch works perfectly.
[buildlog_arduino_new.txt](https://github.com/platformio/platform-espressif8266/files/1908372/buildlog_arduino_new.txt)

ivankravets, 2018-04-20:
Could you reproduce this issue with the latest upstream version? http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

mamama1, 2018-04-20:
Hi

using the latest upstream version it comples and uploads but then only the watchdog triggers all the time:

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

mamama1, 2018-04-23:
Hi

i have tried out the latest update which has been released today (1.7.0, Core 2.4.1). Unfortunately exact the same behaviour as with 1.6.0. Don't know what caused the crash which I posted in my last entry - nevermind.

Content of platformio.ini now:
`[env:nodemcuv2]
platform = espressif8266
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
monitor_baud = 115200
upload_speed = 256000
build_flags = -D CONT_STACKSIZE=8192`

mamama1, 2018-04-23:
i turned on debugging and when compiled with platformio, the output looks like this when querying the tcp server:

> WS:ac
> :rn 18
> :ref 1
> WS:av
> :ref 2
> :ur 2
> :ref 2
> :ur 2
> wlan received cmd type 10, cmd 97
> writing TCP_SC_BRIGHTNESS_STEP_GET...
> :wr 1 1 0
> :wrc 1 1 0
> :wr 1 1 0
> :wrc 1 1 0
> writing TCP_SC_BRIGHTNESS_STEP_GET done.
> :sent 1
> :sent 1
> :rcl
> :abort
> WS:ac
> :rn 18
> 

if I repeat the query (however the server does not respond anymore), i always get this:

> WS:ac
> :rn 18
> 

so it seems that the server is accepting the connection (WS:ac) and then it receives 18 bytes, which is also correct (:rn 18) but then nothing else happens.

when I compile with the arduino ide, it looks like this, many more commands are being sent and received and the server stays responsive (the wlan received cmd type... outputs are my personal debug infos, stating that a command has been received by the tcp server):

> WS:ac
> :ref 1
> WS:av
> :ref 2
> :ur 2
> :ref 2
> :ur 2
> :rn 18
> wlan received cmd type 10, cmd 97
> :wr 1 1 0
> :wrc 1 1 0
> :wr 1 1 0
> :wrc 1 1 0
> :sent 1
> :sent 1
> :rcl
> :abort
> :ur 1
> WS:dis
> :del
> WS:ac
> :ref 1
> WS:av
> :ref 2
> :ur 2
> :ref 2
> :ur 2
> :rn 18
> wlan received cmd type 10, cmd 99
> :wr 1 1 0
> :wrc 1 1 0
> :wr 1 1 0
> :wrc 1 1 0
> :sent 1
> :sent 1
[... 426 lines cut ...]
> :rn 18
> :ref 1
> WS:av
> :ref 2
> :ur 2
> :ref 2
> :ur 2
> wlan received cmd type 10, cmd 155
> :wr 1 1 0
> :wrc 1 1 0
> :wr 1 1 0
> :wrc 1 1 0
> :wr 1 1 0
> :wrc 1 1 0
> :sent 1
> :sent 2
> :rcl
> :abort
> :ur 1
> WS:dis
> :del

any idea what's wrong here? why on earth does the same piece of source code behave differently between arduino and platformio?

thanks

mamama1, 2018-04-23:
it looks like that after :rcl and :abort, the not working sketch simply continues with another connection accept WS:ac and :rn 18 instead of discarding the connection, object or whatever (don't know enough about this, sorry). the working sketch does this, before accepting a new connection (WS:ac):

:ur 1
WS:dis
:del

maybe that's the reason.

ivankravets, 2018-04-25:
Could you try to add `lib_archive = false` to `platformio.ini`:
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
monitor_baud = 115200
upload_speed = 256000
build_flags = -D CONT_STACKSIZE=8192

lib_archive = false
```

mamama1, 2018-04-25:
okay i read a bit about it and personally, I don't do anything like __attribute__((weak)) or #pragma weak in my code. however, I can't speak for the libs I'm using.

mamama1, 2018-04-25:
ok i tested lib_archive = false and as soon as that option is set, neither STA nor AP mode works. it just doesn't find any wifi and in ap mode it doesn't broadcast any ssid.

this is my buildlog:

> [04/25/18 22:26:42] Processing nodemcuv2 (framework: arduino; platform: espressif8266; board: nodemcuv2)
> -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
> SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
> Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF MODES: FINDER(chain) COMPATIBILITY(light)
> Collected 30 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <ESP8266WebServer> v1.0
> |   |-- <ESP8266WiFi> v1.0
> |-- <Bounce2> v2.41
> |-- <Adafruit PWM Servo Driver Library> v1.0.2
> |   |-- <Wire> v1.0
> |-- <Wire> v1.0
> |-- <EEPROM> v1.0
> |-- <RTClib> v1.2.0
> |   |-- <Wire> v1.0
> |-- <ESP8266HTTPUpdateServer> v1.0
> |   |-- <ESP8266WiFi> v1.0
> |   |-- <ESP8266WebServer> v1.0
> |   |   |-- <ESP8266WiFi> v1.0
> |-- <ESP8266WiFi> v1.0
> |-- <IRremoteESP8266> v2.3.2
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\ESP8266WiFi.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\ESP8266WiFiAP.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\ESP8266WiFiScan.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\WiFiClient.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\WiFiClientSecure.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\WiFiServer.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\WiFiServerSecure.cpp.o
> Compiling .pioenvs\nodemcuv2\lib757\ESP8266WiFi\WiFiUdp.cpp.o
> Compiling .pioenvs\nodemcuv2\lib851\ESP8266WebServer\ESP8266WebServer.cpp.o
> Compiling .pioenvs\nodemcuv2\lib851\ESP8266WebServer\ESP8266WebServerSecure.cpp.o
> Compiling .pioenvs\nodemcuv2\lib851\ESP8266WebServer\Parsing.cpp.o
> Compiling .pioenvs\nodemcuv2\lib851\ESP8266WebServer\detail\mimetable.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7d2\Bounce2_ID1106\Bounce2.cpp.o
> Compiling .pioenvs\nodemcuv2\lib18c\Wire\Wire.cpp.o
> Compiling .pioenvs\nodemcuv2\libe67\Adafruit PWM Servo Driver Library_ID30\Adafruit_PWMServoDriver.cpp.o
> Compiling .pioenvs\nodemcuv2\lib625\EEPROM\EEPROM.cpp.o
> Compiling .pioenvs\nodemcuv2\libb17\RTClib_ID83\RTClib.cpp.o
> Compiling .pioenvs\nodemcuv2\lib1a0\ESP8266HTTPUpdateServer\ESP8266HTTPUpdateServer.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\IRrecv.cpp.o
> .piolibdeps\RTClib_ID83\RTClib.cpp: In constructor 'DateTime::DateTime(const char*, const char*)':
> .piolibdeps\RTClib_ID83\RTClib.cpp:136:70: warning: operation on '((DateTime*)this)->DateTime::m' may be undefined [-Wsequence-point]
> case 'J': m = date[1] == 'a' ? 1 : m = date[2] == 'n' ? 6 : 7; break;
> ^
> .piolibdeps\RTClib_ID83\RTClib.cpp: In constructor 'DateTime::DateTime(const __FlashStringHelper*, const __FlashStringHelper*)':
> .piolibdeps\RTClib_ID83\RTClib.cpp:161:70: warning: operation on '((DateTime*)this)->DateTime::m' may be undefined [-Wsequence-point]
> case 'J': m = buff[1] == 'a' ? 1 : m = buff[2] == 'n' ? 6 : 7; break;
> ^
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\IRsend.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\IRtimer.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\IRutils.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Aiwa.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Argo.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Carrier.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Coolix.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Daikin.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Denon.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Dish.cpp.o
> Compiling .pioenvs\nodemcuv2\lib7ee\IRremoteESP8266_ID1089\ir_Fujitsu.cpp.o
[... 82 lines cut ...]
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\spiffs\spiffs_check.c.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\spiffs\spiffs_gc.c.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\spiffs\spiffs_hydrogen.c.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\spiffs\spiffs_nucleus.c.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\spiffs_api.cpp.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\spiffs_hal.cpp.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\time.c.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\uart.c.o
> Compiling .pioenvs\nodemcuv2\FrameworkArduino\umm_malloc\umm_malloc.c.o
> Archiving .pioenvs\nodemcuv2\libFrameworkArduino.a
> Linking .pioenvs\nodemcuv2\firmware.elf
> Checking program size
> text       data     bss     dec     hex filename
> 395388     9596   35296  440280   6b7d8 .pioenvs\nodemcuv2\firmware.elf
> Building .pioenvs\nodemcuv2\firmware.bin
> Configuring upload protocol...
> Looking for upload port...
> Auto-detected: COM10
[... cut ...]
> Uploading 409136 bytes from .pioenvs\nodemcuv2\firmware.bin to flash at 0x00000000
> ................................................................................ [ 20% ]
> ................................................................................ [ 40% ]
> ................................................................................ [ 60% ]
> ................................................................................ [ 80% ]
> ................................................................................ [ 100% ]

TD-er, 2018-05-04:
I am experiencing something similar while building with ESPeasy, using Platformio 1.7.0
It looks like when there are wifi issues, a rebuild of the same code may suddenly work.
This is quite frustrating, since I have been bug-hunting these wifi issues for a long time now and the issues reported by users are very hard to explain given the code and the changes made.
Only the last few days I've noticed this strange behavior that rebuilding it twice may fix these issues.

So it seems there is something wrong with the linking process?
I really can't tell for sure, but it is rather frustrating to see user being very happy about fixes and then new reports based on nightly builds with exactly the same code and new issues on new builds when nothing related to wifi has changed, or the opposite when things start working on a new build.

davidmpye, 2018-05-24:
This may or may not be related, but I have a bunch of generic ESP12E boards, and cannot even get simple code compiled in platformio to run on then, If I compile the same code with arduino 1.6.11 it works fine.   This has *definitely* changed, as I normally use platformio to do all my dev work.

Example:
`#include <Arduino.h>

void setup() {
  pinMode(2,OUTPUT);
}

void loop() {
digitalWrite(2, HIGH);
delay(100);
digitalWrite(2, HIGH);
delay(100);
digitalWrite(2, LOW);
delay(1000);
}`

If I compile with the arduino IDE, works as expected.  If I upload via Platformio, wdt just gets triggered, and the code doesn't run.

I've tried editing platformio.ini to mimic the settings I have in the arduino IDE :+1: 
`[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
board_f_cpu = 80000000L
board_f_flash = 40000000L
board_flash_mode = dio
build_flags = -Wl,-Teagle.flash.4m.ld
upload_resetmethod = ck`

but still no joy.

TD-er, 2018-05-24:
Have you tried building it twice before upload?

TD-er, 2018-05-24:
There is something very funky going on with the PlatformIO build process.
I am not sure when it started, but I know I lost really a lot of hours (way over a month times my sparse free time to work on ESPeasy) trying to find lots of reported issues/crashes/etc that I could not reproduce.

You have to make sure to close and re-open Atom with PlatformIO plugin when changing anything in the platformio.ini file or any file outside the editor (e.g. rebase from git) before building again and when you change Arduino core libraries (e.g. 2.3.0 => 2.4.x) you have to build it at least twice to make sure it is working.
And still then it is possible the built binaries are not working as intended.
Often I see issues with WiFi, especially since working with 2.4.x (Platformio 1.6.0 or higher).

It is really bothering me too, since there is no guarantee the nightly builds we make are working and even worse, among builds in the same run things can get messy.
And we're building it twice already.

mamama1, 2018-05-24:
Can I do anything to help troubleshooting or finding the root cause?
Unfortunately my skills aren't too good so all I can do is try things, post logs, give feedback - but maybe that'd help?

ascillato, 2018-05-29:
I'm experiencing the same issue. Any news?

ivankravets, 2018-05-29:
@ascillato  @mamama1 could you provide your `platformio.ini` files? Maybe, simple sketches which do not work? Thanks!

adamrabbani, 2018-06-27:
me too.... upload successfully but when i run doesn't happen.
when i upload on IDE.. and i run, output is good..

ascillato, 2018-09-08:
Hi,

@ivankravets 

We found with @andrethomas and @arendst compiling Sonoff-Tasmota that if you make a **compile** and you have files in the folder .pioenvs, and that you have changed just few things in your code (like a #define), platformio compiles a bin with a misaligment that produces wifi issues.

However, If you click **Rebuild C/C++** before every compiling/upload, platformio compiles with no problem.

This misaligment is sometimes random, but with the **rebuild c/c++** click before compilation always produce correct binary file.

Can be added an option to rebuild c/c++ before compilation?

Thanks.

TD-er, 2018-09-08:
Just curious, what do you mean with "Rebuild C/C++" ?
I can only see the "Rebuild C/C++ Project Index (Autocomplete, Linter)

I am really interested in such a feature too, since way too much time has been lost on wrong builds or error-reports of users claiming wifi issues on self-made builds.
For the nightly builds I've been building all things twice and that reduced the number of "funky builds" significantly, but not to zero. (ESPeasy by the way)

ascillato, 2018-09-08:
Yes, exactly that option I meant, to click on **Rebuild C/C++ Project Index (Autocomplete, Linter)**

ascillato, 2018-09-08:
Also, if you delete the .pioenvs folder before compile, the misaligment issue is not there.

ascillato, 2018-09-08:
Because of this, would be super handy to have an option for platformio.ini to do a **Rebuild C/C++ before compiling**

TD-er, 2018-09-09:
You could ofcourse add a pre-build script to throw that directory away, but then the build times increase.
It isn't a big deal for nightly builds, but during development the building is already a rather time consuming part.  (not to mention the restart time of Atom when switching branches, but that's another issue)

arendst, 2018-09-09:
Guys, I see no option **Rebuild C/C++ before compiling** probably because I'm using VSC on windows that does have an option **Rebuild IntelliSense Index**. As the original feeling was that on windows the alignment problem did not occur it could be because I see this IntelliSense Rebuild going on almost every time before or after any compile.

I might be cloudy-ing this discussion comparing two unrelated rebuilds.

ivankravets, 2018-09-09:
Rebuilding Project IntelliSense Index is not related to this issue. It does the same as you do when try to build project. 

If you need to remove previous build artifacts, just run Clean task/target.

arendst, 2018-09-09:
@ivankravets OK thanks. But where in VSC on windows is this **Rebuild C/C++ before compiling** option?

ivankravets, 2018-09-09:
Right side bar > PlatformIO icon > Project Tasks > Rebuid IntelliSense Index. 

This tasks just rebuilds index DB for VSCode C/C++ extension to provide code completion and code linting

gmeurant, 2018-10-17:
Same problem here, using a D1 mini lite (ESP8266) and a Wifi Server in a converted Arduino project : it never connects to my home wifi router when compiled and built with PIO, but works fine with Arduino IDE. It really seems to be wifi related, or at least wifi library related.
My included headers for information:
`#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <time.h>
#include <AutoConnect.h>
`

ivankravets, 2018-10-18:
Could someone provide us with a simple project which works with Arduino IDE and does not work with PlatformIO?
Thanks!

franzzi, 2018-10-26:
I have the same issue with an example for AsyncWifiManager. Same code works with Arduino IDE without problems, but won't work with PlatformIO. What I can also see is that binary has different size. (Arduino IDE 289KB, PlatformIO 297KB)
Arduino IDE version: 1.8.7
PlatformIO version: 1.0.4 Core: 3.6.1rc6
Here's the example mentioned before:

`#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
#include <WiFi.h>
#endif

//needed for library
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>         //https://github.com/tzapu/WiFiManager

AsyncWebServer server(80);
DNSServer dns;

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    AsyncWiFiManager wifiManager(&server,&dns);
    //reset saved settings
    //wifiManager.resetSettings();
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
}`
BR

ivankravets, 2018-10-29:
@valeros could you help with the latest example? Does it work for you?

valeros, 2018-10-29:
Hi @franzzi ! Could you explain what exactly doesn't work? Just tried your example and looks like it works as expected with my nodemcu board.
```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.0.105
connected...yeey :)
```

franzzi, 2018-10-30:
Hi @valeros, please clear your flash before you load the application. In this case there are no saved values for IP address available and WifiManager will start an Access Point. If everything works you can reach WifiManager via IP 192.168.4.1. 
This works for me just if I use Arduino IDE for compilation.
If I use PlatformIO for compilation job, i can't reach access point with 192.168.4.1.
I'm using the same libraries in both cases, but the size of binaries is different. I guess that some compiler flags are different but I'm not sure.

valeros, 2018-10-31:
@franzzi Could you please try to add `PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY` flag to your `platformio.ini` ? Something like this:
```
...
build_flags = -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
```

valeros, 2018-10-31:
@franzzi Actually, I'm not sure that this problem is only related to PlatformIO. Your example won't work in Arduino IDE as well if you choose `lwIP v1.4 Higher Bandwith` as your lwIP variant (which we use as default in our build script). I've not looked into the problem closely enough to jump to conclusions, but there is something wrong with this combination of libraries and lwIP. Aside from that, there have been some changes in lwIP libraries in the official Arduino framework repository, so probably we will ask them which lwIP library we should use by default.

d-a-v, 2018-11-01:
> so probably we will ask them which lwIP library we should use by default.

> `build_flags = -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY`

This is indeed the default choice in the arduino IDE (core 2.4.0 and after).
It selects lwIP-v2 with TCP-MSS=536.
You can also use `build_flags = DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH` (TCP-MSS=1460).

It will be a loss of time to find out why lwIP-1.4 does not work. They are supposed to be interchangeable.
lwIP-v2 came to fix lwIP-v1.4 issues although this port took long time to stabilize.
lwIP-v2 port is considered stable in core-2.4.2, and performances / tcp stability have improved in core-stage (git version).

franzzi, 2018-11-01:
@valeros & @d-a-v 
Thx for your answers!
BR franzzi

ivankravets, 2018-11-02:
@d-a-v Thanks! I've just updated our package in CDN for 2.4.2, it will use `PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY` profile automatically if another is not specified.

Also, see PR for "stage" version https://github.com/esp8266/Arduino/pull/5305

ivankravets, 2018-11-02:
Please update packages via `pio update`.

closed 2018-11-02 by ivankravets