platformio/platform-espressif8266 #85: Update native SDK for ESP8266 to 2.0.0

Droncz, 2018-06-11:
Hello, any chance of updating the build environment, so it works with a new version of the native RTOS based SDK (https://github.com/espressif/ESP8266_RTOS_SDK)?
This kinda relates to #79 so You would actually kill two birds with one stone ;-)

Thanks.

vshymanskyy, 2018-08-29:
There's already v 3.0.0 but it consists of major component refactoring

Random90, 2018-12-15:
Update would be awesome. Current version is based on community package written few years ago, and nobody updated it yet. Is ESP8266 slowly replaced by ESP32?

bwssytems, 2019-01-08:
Would really like to have V3.0.0 of the RTOS Framework since the code written for ESP-IDF will work with this. Sometimes you need an 8266 instead of the 32

maxgerhardt, 2019-02-10:
*Push!* This seems to occur [over and over again](https://community.platformio.org/t/update-to-esp8266-rtos-sdk/6485); Espresif RTOS SDK is already at 3.1 now.

WallaceWilliam, 2019-03-02:
try it 
https://github.com/WallaceWilliam/framework-esp8266-rtos-sdk-idf-platformio