platformio/platform-espressif8266 #87: Unable to compile v

mauriciojost, 2018-06-23:
When using espressif8266 @ 1.7.2 I can't compile my project. 
However when forcing to espressif8266 @ 1.7.1 or espressif8266 @ 1.7.0 everything seems to work correctly. 

See logs: 

```
+ platformio run
********************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
********************************************************************************

[Sat Jun 23 11:08:38 2018] Processing main (platform: espressif8266; build_flags: -D LOG_LEVEL=0 -D YES_DEBUG -Og -ggdb, -D WIFI_SSID_INIT=\"wifi\", -D WIFI_PASSWORD_INIT=\"pass\", -D WIFI_SSID_STEADY=\"wifi\", -D WIFI_PASSWORD_STEADY=\"pass\", -D ENCRYPT_KEY=\"11112233445566778899aabbccddeeff\", -D DWEET_IO_API_TOKEN=\"token\", -D TIMEZONE_DB_KEY=\"key\", -D TIMEZONE_DB_ZONE=\"Europe/Paris\", -D MAX_LOG_MSG_LENGTH=128, -D OVERRUN_THRESHOLD_SECS=360, -D F_INCR_MIN=-1, -D F_INCR_MAX=1, -D PERIOD_SEC=60, -D MAX_EFF_STR_LENGTH=64, -D SERVO0_INVERTED=false, -D SERVO1_INVERTED=true, -D SERVO0_BASE_DEGREES=60, -D SERVO1_BASE_DEGREES=60, -D SERVO0_RANGE_DEGREES=100, -D SERVO1_RANGE_DEGREES=100, -D SECS_CYCLE_FACTOR_DEFAULT=PERIOD_SEC -D DEVICE_NAME=\"dev1\"; board: esp12e; framework: arduino)
--------------------------------------------------------------------------------
PlatformManager: Installing espressif8266
Downloading
Downloading
Unpacking
Unpacking
espressif8266 @ 1.7.2 has been successfully installed!
The platform 'espressif8266' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Downloading
Downloading
Unpacking
Unpacking
PackageManager: Installing tool-esptool @ <2
Downloading
Downloading
Unpacking
Unpacking
PackageManager: Installing framework-arduinoespressif8266 @ ~1.20401.1
Downloading
Downloading
Unpacking
Unpacking
CorePackageManager: Installing tool-scons @ ~3.20501.2
Downloading
Downloading
Unpacking
Unpacking
tool-scons @ 3.20501.2 has been successfully installed!
$ docker exec a249eeb322c74d810f5c078508c11ff460aabcf51b9485e2970211665e60aebc env SSH_AGENT_PID=15 SSH_AUTH_SOCK=/tmp/ssh-PbMB5qyYNZGz/agent.10 ssh-agent -k
unset SSH_AUTH_SOCK;
unset SSH_AGENT_PID;
echo Agent pid 15 killed;
[ssh-agent] Stopped.
Verbose mode can be enabled via `-v, --verbose` option
TypeError: __init__() takes exactly 2 arguments (1 given):
File "/usr/local/lib/python2.7/site-packages/platformio/builder/main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "/var/jenkins_home/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/var/jenkins_home/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/var/jenkins_home/.platformio/platforms/espressif8266/builder/main.py", line 49:
@util.memoized()
========================== [ERROR] Took 9.17 seconds ==========================
[Pipeline] }
```

vs 

```
[Pipeline] sh
+ platformio run
[... 213 lines cut ...]
Indexing .pioenvs/main/libc35/libSPI.a
Compiling .pioenvs/main/lib791/EEPROM/EEPROM.cpp.o
Archiving .pioenvs/main/libf91/libServo.a
Indexing .pioenvs/main/libf91/libServo.a
Compiling .pioenvs/main/libf41/ESP8266mDNS/ESP8266mDNS.cpp.o
Archiving .pioenvs/main/lib791/libEEPROM.a
Indexing .pioenvs/main/lib791/libEEPROM.a
Compiling .pioenvs/main/libaed/ArduinoOTA/ArduinoOTA.cpp.o
Archiving .pioenvs/main/libaed/libArduinoOTA.a
Indexing .pioenvs/main/libaed/libArduinoOTA.a
Archiving .pioenvs/main/libf41/libESP8266mDNS.a
Indexing .pioenvs/main/libf41/libESP8266mDNS.a
Linking .pioenvs/main/firmware.elf
Calculating size .pioenvs/main/firmware.elf
Building .pioenvs/main/firmware.bin
text	   data	    bss	    dec	    hex	filename
377504	  15592	  31104	 424200	  67908	.pioenvs/main/firmware.elf
========================= [SUCCESS] Took 36.21 seconds =========================

********************************************************************************
There is a new version 3.5.3 of PlatformIO available.
Please upgrade it via `platformio upgrade` or `pip install -U platformio` command.
Changes: http://docs.platformio.org/en/latest/history.html
********************************************************************************
```

ivankravets, 2018-06-23:
The problem is linked with old version of PIO Core on your machine. `pio upgrade` will fix it

closed 2018-06-23 by ivankravets

mauriciojost, 2018-06-24:
Good to know, thanks for your help. 

Mauricio