platformio/platform-espressif8266 #90: ESP8266 remote error eagle.flash.4m1m.ld

jasonbronson, 2018-07-23:
I installed core on my remote linux machine and it runs fine but when I execute and upload from another location I get this error.
I’ve tried to remove .platformio directory and reinitialize but no luck.
When I run “platformio update” it shows esp8266 installed

Error: Could not find ‘eagle.flash.4m1m.ld’ LD script in LDPATH ‘/home/pi/.platformio/remote/projects/Freezer Alarm-d4a3c214ac103c256adf417b8baf78f22b419df2/.pioenvs/huzzah’

full debug log here
https://pastebin.com/EbxAu0wW

Also attempting to run 
`pio remote run --force-remote command`

ivankravets, 2018-07-30:
Fixed in https://github.com/platformio/platformio-core/commit/ae964fa7292439da8899cac54478591791b8ccd9

Please run `pio upgrade --dev` on the both sides. Does it work now?

closed 2018-07-30 by ivankravets

a-x-, 2018-12-13:
It's broken for me still

a-x-, 2018-12-13:
pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 3.6.3

```
> Executing task: platformio run --target upload <

Processing esp8266_12 (platform: espressif8266; board: esp12e)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Error: Could not find 'eagle.flash.4m1m.ld' LD script in LDPATH '/Users/mxtnr/Documents/Arduino/LightController2/.pioenvs/esp8266_12 /Users/mxtnr/.platformio/packages/sdk-esp8266/lib /Users/mxtnr/.platformio/packages/sdk-esp8266/ld'
Error: Could not find 'eagle.flash.4m1m.ld' LD script in LDPATH '/Users/mxtnr/Documents/Arduino/LightController2/.pioenvs/esp8266_12 /Users/mxtnr/.platformio/packages/sdk-esp8266/lib /Users/mxtnr/.platformio/packages/sdk-esp8266/ld'
*** [.pioenvs/esp8266_12/eagle.flash.bin] Error 1
===================================================================== [ERROR] Took 0.51 seconds =====================================================================
The terminal process terminated with exit code: 1
```

mikewebb70, 2018-12-13:
-  upgraded to Atom 1.33.0 Platformio Home 2.0.0·Core 3.6.3
-  added the following to my platform.ini
  [env:nodemcuv2]
  platform = espressif8266
  board = nodemcuv2
-  did a pio upgrade --dev 
- Build with PIO Build (nodemcuv2)

Same errors as others:

platform run --environment nodemcuv2
```
Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2)

--------------------------------------------------------------------------------
Warning! `board_flash_mode` option is deprecated and will be removed in the next release! Please use `bo
ard_build.flash_mode` instead.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.6
|-- <NeoPixelBus> 2.2.9
|-- <IRremoteESP8266> 2.2.1
Error: Could not find 'eagle.flash.1m0.ld' LD script in LDPATH '/Users/mike/Documents/PlatformIO/Project
s/Sonoff-Tasmota-6.3.0 2/.pioenvs/nodemcuv2 /Users/mike/.platformio/packages/sdk-esp8266/lib /Users/mike
/.platformio/packages/sdk-esp8266/ld'
Error: Could not find 'eagle.flash.1m0.ld' LD script in LDPATH '/Users/mike/Documents/PlatformIO/Project
s/Sonoff-Tasmota-6.3.0 2/.pioenvs/nodemcuv2 /Users/mike/.platformio/packages/sdk-esp8266/lib /Users/mike
/.platformio/packages/sdk-esp8266/ld'
*** [.pioenvs/nodemcuv2/eagle.flash.bin] Error 1
========================== [ERROR] Took 1.60 seconds ==========================
```

ivankravets, 2018-12-13:
Could you provide a simple project to reproduce this issue?

mikewebb70, 2018-12-14:
Ivan, looking at the issue, mine is not related to remote flashing.  Direct usb/seriall connection to nodemcu lolin v3 (nodemcu v1.0 esp-12e) uploading tasmota. Everything builds successfully in targeted mode but getting eagle.flash.1,0.ld errors on upload.

I looked at what libraries are in https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld to see if it is there.  There is an eagle.flash.1.ld  but no eagle.flash.1,0.ld.   Thats as far as I got as I was in hurry so I fired up Arduino IDE and flashed.

ivankravets, 2018-12-14:
@mikewebb70 fixed in https://github.com/platformio/platformio-docs/commit/925dd823b1bc2fc24774fee25ed2b42a5fe0c321

So, I see that Espressif team has changed the names for a few LD scripts in upcoming 2.5.0 release. We have removed a list with available LD scripts. The actual information is here https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld

Please switch between `master` or release branches for the actual information.

mikewebb70, 2018-12-15:
Mark my issue as closed,  I'm taking this over to Tasmota crew.  Thanks

drnic, 2019-03-04:
@mikewebb70 can you link to where you moved your issue and had it resolved?

ivankravets, 2019-03-04:
See https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

mikewebb70, 2019-03-04:
@drnic my issue stemmed from the build flags in the default platformio.ini that came with tasmota.  Good for it’s intended target of SONOFF devices but overrided and caused my platformio settings for nodemcu.  Ivan’s link above explains perfectly how to solve such conflicts.  Like I said before, I was having a newbie end user issue.  Thanks Ivan.

drnic, 2019-03-04:
@mikewebb70 thanks; turns out my own similar looking issue was a regression in espressif8266 2.0.0/2.0.1; apparently fixed on master; I got around it by downgrading to 1.8.0 (for anyone else finding this thread before 2.0.2 comes out)

ivankravets, 2019-03-05:
Yes, we had a bug in 2.0.0 release of ESP8266 dev platform. Please update to the latest 2.0.2