platformio/platform-espressif8266 #93: Plan for releasing core 2.4.2?

Frogmore42, 2018-08-08:
I know that I can use the latest git version.  It would be really nice if there was a similar/easy way to use any of the released versions from https://github.com/esp8266/Arduino/releases

2.4.2 was released 5 days ago.  It looks like it took about a month for 2.4.1 to make it to a platformio release, based on: https://github.com/platformio/platform-espressif8266/issues/76

 Should I expect something similar for 2.4.2?

ivankravets, 2018-08-08:
We will release it soon. Does 2.4.2 work well for you?

Frogmore42, 2018-08-08:
I haven't tried it yet, but I have heard it is much better for some stability issues that are in 2.4.1.  Using it from the Arduino environment is possible, but that environment is much less nice for complex projects.

commit b8b5a24d (2018-08-11): Update Arduino core to 2.4.2 // Resolve #93
commit ac1e0c84 (2018-08-11): Merge branch 'develop' into feature/stage

* develop:
  Fix simba framework
  Add new boards: Amperka WiFi Slot, WiFiduino, SeeedStudio Wio Link, XinaBox CW01
  Update Arduino core to 2.4.2 // Resolve #93
  Move common flags to framework build script
  Rename monitor_baud to monitor_speed
  Bump version to 1.7.3

# Conflicts:
#	boards/wifiduino.json
#	platform.json
closed 2018-08-11 by ivankravets