platformio/platform-espressif8266 #98: Arduino Core 2.4.2 WPS requires build flag

dalbert2, 2018-08-31:
For release 2.4.2 of the Arduino Core, WPS functionality has been removed by default in favor of a 4K memory optimization.  To use WPS, you must include the following build flag: -D NO_EXTRA_4K_HEAP

This should probably be added to the documentation

tve, 2018-09-03:
Got a link to any docs explaining what changed?

dalbert2, 2018-09-03:
From the release notes:

https://github.com/esp8266/Arduino/releases

See 11th bullet under Core:

  * Move continuation stack from .bss onto sys stack, saving 4kB of RAM,
    if WPS is not needed (#4622
    <https://github.com/esp8266/Arduino/pull/4622>)

Regards,
Dave

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus