platformio/platform-espressif8266 #100: Update Arduino frameworks in devel or stage branches

CRCinAU, 2018-10-14:
There has been a ton of changes to the Arduino framework here:
https://github.com/esp8266/Arduino/commits/master

This includes possible fixes for the cause of failing OTA updates, httpClient fixes, and more.

Is there any chance of either automating the import of these changes to a rolling branch to be used within PIO, or some other method used to keep things somewhat up to date?

CRCinAU, 2018-10-16:
See this Issue for more details:
https://github.com/esp8266/Arduino/issues/5023

CRCinAU, 2018-10-22:
@ivankravets ping?

ivankravets, 2018-10-22:
Have you checked our docs? You need upstream version http://docs.platformio.org/en/latest/platforms/espressif8266.html

closed 2018-10-22 by ivankravets

CRCinAU, 2018-10-22:
The docs say to use the following for 'upstream':
```
platform = https://github.com/platformio/platform-espressif8266.git
```

This doesn't seem to match what I see in https://github.com/esp8266/Arduino

I've asked in multiple places, the consensus seems to be that the esp8266/Arduino repo needs to be pulled into the PIO one somehow?

ivankravets, 2018-10-22:
> This doesn't seem to match what I see in https://github.com/esp8266/Arduino

How did you check it?

CRCinAU, 2018-10-22:
When I do a build, ESP.getFullVersion() returns:
```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
```

In theory, lwIP should be 2.1.something if tracking master on the Arduino repo.

ivankravets, 2018-10-22:
Maybe, something is not updated in their upstream version. I mean, they do this replacement on release. You can ignore the result from ESP.getFullVersion().

`platform = https://github.com/platformio/platform-espressif8266.git` will use the latest Arduino ESP8266 Core. You can check it in `~/.platformio/packages`

CRCinAU, 2018-10-23:
Looking at this commit here:
https://github.com/esp8266/Arduino/commit/a1e59e9c014c907c008e96c9a3f1b7fc731adedb

The second chunk adds "LWIP_FEATURES". From the 'upstream' version configured in platformio.ini, I get no matches in tools/platformio-build.py

I also get no matches for:
`grep LWIP_NO_CTYPE tools/sdk/lwip2/include/arch/cc.h`

The versioning also differs:
Upstream: #define LWIP_HASH_STR "STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82"
The commit: #define LWIP_HASH_STR "STABLE-2_1_0_RELEASE/glue:arduino-2.4.2-13-g80224f0"

ivankravets, 2018-10-23:
Oh, sorry! :( This my mistake. You need a staging version. See http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

CRCinAU, 2018-10-23:
Ahhh awesome - thanks.

I can confirm that this fixes all my OTA update problems with Cisco access points as well.