platformio/platform-espressif8266 #104: how to add patch on top of Arduino framework?

uzi18, 2018-10-21:
Is it possible to use upstream patch before project build?

TD-er, 2018-12-11:
@uzi18 
I was looking into this [custom platform documentation](https://docs.platformio.org/en/latest/platforms/creating_platform.html)
Maybe we can make our own set of platform definitions, which already contain some patches, etc.

Only thing is, I don't know what part should be the 'arduinoespressif8266' package.
See my other issue: #109

TD-er, 2019-01-05:
@uzi18 See the current platformio.ini of ESPeasy.
It is now using pre-patched version of Puya patch, which is back-ported from core 2.5.0

In short, I had to fetch the platform package, run the patch manually and make a tar.gz from it.
Also the package.json file has to be changed to allow the "-puya" postfix in the version.
These files I had to upload to my server and generate a SHA for it.
This is all included in the manifest.json I put next to the files.

For example core 2.4.1:
```
[core_2_4_1]
platform                  = https://github.com/TD-er/platform-espressif8266.git#patch/v1.7.3_Puya
```

That's a branch on the fork of this repo on my account.
In there I made a single commit: https://github.com/TD-er/platform-espressif8266/commit/750760555e9bf00eaccf1395707f9b669b1ade3d
to set the repository to my fork and add a packageRepository location to my server. (manifest.json file)
That's the only repository able to serve `"version": "1.20401.3-puya"` which is the one we're requesting in our platform.json file.

So it's a lot of work, and also it will not keep in sync when @ivankravets will release a new version.
For older core libs it may be worth it to add something which does not need any patching in future core libs.

uzi18, 2019-01-05:
ok will check this out