platformio/platform-espressif8266 #105: Linker fails after upgrading to 1.8.0

vanniaz, 2018-10-26:
I upgraded the Espressif 8266 platform from 1.7.0 to 1.8.0, and now the project build fails at the linking stage with this error:
`Linking .pioenvs\esp_wroom_02\firmware.elf
.pioenvs\esp_wroom_02\src\main.cpp.o:(.text.loop+0x48): undefined reference to `g_cont'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\esp_wroom_02\firmware.elf] Error 1`

So I downgraded to 1.7.0 and it builds correctly.
What can I check?

Thanks for your help

peastone, 2018-11-01:
This happens also for me: .pioenvs/esp01_1m/src/WIFIOnOff.ino.cpp.o:(.text._Z10performWPSv+0x10): undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

I downgraded to 1.7.0.

ivankravets, 2018-11-02:
Do you see the same issue with Arduino IDE and Core 2.4.2? Maybe, you need to update your code?

ivankravets, 2018-11-02:
Also, please take a look at https://github.com/platformio/platform-espressif8266/issues/98

platformio.ini
```
[env:myenv]
board = ...
build_flags = -D NO_EXTRA_4K_HEAP
```

vanniaz, 2018-11-02:
Yes, the -D NO_EXTRA_4K_HEAP build flag did the trick!
But I am not using WPS, so why is this extra 4k heap needed?

I checked with Arduino IDE (I was not using it so I had to install it) and the behaviour is the same, so this is not a Platformio related issue. I should probably submit an help request to esp8266/Arduino.
Thanks
Giovanni

peastone, 2018-11-02:
Adding the build_flags did the trick! Thank you @ivankravets!

ivankravets, 2018-11-03:
Please move this issue to https://github.com/esp8266/Arduino/issues

closed 2018-11-03 by ivankravets