platformio/platform-espressif8266 #107: mbedos 5.10.x with ESP8266 fails to build

mburumaxwell, 2018-11-04:
I am facing an error using ESP8266 in mbed on platformio.
Steps to reproduce:

1. create project based on any example (best case is _mbed-rtos-ethernet_)
2. build project (it succeeds)
3. Add `#include "ESP8266Interface.h"`
4. build project (it fails)

Errors are too many but they all point towards missing header files like WiFiAccessPoint.h, WiFiInterface .h etc.
I beleive this is probably an issue with mbedos itself but on the mbed compiler it works. Is there a way to resolve this issue?

ivankravets, 2018-11-04:
What is your `platformio.ini`?

mburumaxwell, 2018-11-04:
Hi @ivankravets , here is the platformio.ini file

```
[env:nucleo_f767zi]
platform = ststm32
framework = mbed
board = nucleo_f767zi
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```

Just to clarify, this happens when ESP8266 is used as a slave module to another microcontroller running mbed-os 5.10.x. It is misplaced in the ESP8266 platform. It also does not matter whether the board has a microcontroller by STMicroelectronics or NXP.