platformio/platform-espressif8266 #109: How to run beta version of esp8266/Arduino

TD-er, 2018-12-11:
Every now and then one of the supported platforms will release a test version and then a new version.
Updates to a released new version are done quite fast, but (understandably) the beta versions are not added.
For example the [esp8266/Arduino](https://github.com/esp8266/Arduino/tree/2.5.0-beta1) made a beta available for the 2.5.0 version.

What has to be done to make a test build using PlatformIO, based on a tag or branch of such a platform?
This will ease the testing phase for such a new platform, if we can add it ourselves in some kind of test setup.

Zoptune, 2018-12-16:
Hi,
I was also interested in using the 2.5.0-dev version.
I downloaded the zip file from the github link you mentioned, then I extracted it in `.platformio/packages/framework-arduinoespressif8266@2.5.0-dev`
I also changed the framework version in the `.platformio/platforms/espressif8266/platform.json` :
`"framework-arduinoespressif8266": {
      "type": "framework",
      "optional": true,
      "version": "~2.5.0-dev"
    },`

Don't know if it's the right way to do it but it works.

CyanoFresh, 2019-08-07:
definitely need a guide on this topic