platformio/platform-espressif8266 #110: PackageManager reports unknown package if it is unable to download the package, even if it actually exists

mincrmatt12, 2018-12-14:
### Configuration

**Operating system**: Ubuntu 18.04.1 LTS

**PlatformIO Version** (`platformio --version`): 3.6.3

### Description of problem

While attempting to install some packages on a filtered network (specifically, one that blocks the raw.githubusercontent.com domain), PackageManager reported

```
Error: Detected unknown package 'framework-esp8266-rtos-sdk
```

despite the fact that this package does in fact exist.

#### Steps to Reproduce

1. Artificially block the raw.githubusercontent.com domain
2. Attempt to install the `framework-esp8266-rtos-sdk` package

### Actual Results

```
PlatformManager: Installing framework-esp8266-rtos-sdk
Error: Detected unknown package 'framework-esp8266-rtos-sdk
```

### Expected Results

```
PlatformManager: Installing framework-esp8266-rtos-sdk
Error: Unable to download 'framework-esp8266-rtos-sdk
```
(or some message to indicate the issue is not a missing package, rather the inability to download it)

mincrmatt12, 2018-12-14:
@ivankravets Are you sure this should be here? The esp8266 package was used as an example, the bug occurs for any package.

ivankravets, 2018-12-14:
Could you try to run in PIO IDE Terminal
```
pio run 
```

Do you see more detailed info why a package is not found?

mincrmatt12, 2018-12-14:
I am able to access the first two, but the last 4 are blocked.

ivankravets, 2018-12-14:
Do you have `python-requests` package in a system? Or TLS/SSL libs? You need update them.

mincrmatt12, 2018-12-14:
To clarify, I'm not asking for help downloading the packages - I'm fully aware my network filters requests to githubusercontent. I'm just saying that the error reported is very misleading.

ivankravets, 2018-12-14:
The idea of package mirrors to ignore off-lined and search for another. See https://github.com/platformio/platformio-core/blob/develop/platformio/managers/package.py#L66

So, very often some mirrors are down and other online. In this case, we don't show "misleading" warnings. The problem is really with only this package `framework-esp8266-rtos-sdk`, it is not mirrored and available only via Github manifest. I'll think how to resolve this issue better.

mincrmatt12, 2018-12-14:
Alright, that makes sense. Thanks.