platformio/platform-espressif8266 #114: Images seems to always be compiled for 4MB flash

buzzy, 2019-01-10:
I have created a new project in PlatformIO. I picked the board "Wemos D1 Mini Lite", which has 1MB of flash. When compiling the code, I see that it has been picked correctly:

CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_lite.html
PLATFORM: Espressif 8266 > WeMos D1 mini Lite
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)

However, when trying to flash this image to the esp8266 via HTTPUpdater, it keeps being identified as a 4MB target bin file:

Update error: ERROR[8]: Flash config wrong real: 1048576 IDE: 4194304

Is this a bug or am I missing something?

debsahu, 2019-01-17:
Default ld file has 4MB target 3MB spiffs. see comments here (as [documentation suggests](https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size)):
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.4m.ld
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.4m1m.ld

buzzy, 2019-01-17:
Not correct. Default ld file will be picked based on what board you have. But regardless, I have already tried with manual setting the ld-file to 1M

build_flags = -Wl,-Teagle.flash.4m.ld

Makes no difference.

debsahu, 2019-01-17:
Use the staging version. And use downloaded version of https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.1m.ld

pfeerick, 2019-01-28:
What impact does the `f_flash": "40000000L",` have on the build process? Does it download more FLASH memory for the ESP8266? :stuck_out_tongue: https://github.com/platformio/platform-espressif8266/blob/develop/boards/d1_mini_lite.json#L6

ivankravets, 2019-02-18:
Does it work with the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif8266.html#upstream?

ivankravets, 2019-02-24:
Please reopen if you still need help

closed 2019-02-24 by ivankravets

RouNNdeL, 2019-06-07:
Has anyone managed to fix this error? I've tried:
- Overriding the `board_upload.maximum_size = 1048576` 
- Manually setting the ld script `board_build.ldscript = eagle.flash.1m.ld` 
- Using custom build flags `-Teagle.flash.1m0.ld`

It used to work a few months back without an issue

ivankravets, 2019-06-07:
> -Teagle.flash.1m0.ld

Please use `build_flags = -Wl,-Teagle.flash.1m0.ld`

See http://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

RouNNdeL, 2019-06-07:
I managed to fix it by uploading a newer library version over Serial

buzzy, 2019-06-07:
That's what I did in the end as well. Uploading through WiFi simply does not work, even when using those build_flags.

MarekUniq, 2019-08-05:
Have anybody found a workaround/solution?
I think I hit the same issue.

OTA Upgrade ends with message:
> 2019-08-05T21:44:39
> Failed Update error: ERROR[8]: Flash config wrong real: 1048576 IDE: 4194304

Compiling commands:
> platformio run -e sonoff
> platformio run -e sonoff-minimal

Log of compile shows correct size, "1MB Flash":
> Processing sonoff ( framework: arduino; platform: espressif8266@~2.2.2; board: esp01_1m)
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
> PLATFORM: Espressif 8266 2.2.2 > Espressif Generic ESP8266 ESP-01 1M
> HARDWARE: ESP8266 80MHz, 80KB RAM, **1MB Flash**
> PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)