platformio/platform-espressif8266 #116: tool-scons not available for darwin_x86_64

OttoWinter, 2019-01-15:
Recently did a `platformio platform update` for my dev environment on a Macbook. Now I get this error that tool-scons cannot be found for my system architecture:

```
Error: Could not find a version that satisfies the requirement '~3.30100.0' for your system 'darwin_x86_64'
```

I also cannot find a matching version on bintray: https://bintray.com/platformio/dl-packages/tool-scons#files

(Note this is probably a platformio-wide issue, since scons is used by the entire system, but posting it here as for some reason it only shows up for ESP8266 for me)

Platformio.ini:

```ini
[env:test8266]
platform = espressif8266@1.8.0
framework = arduino
board = d1_mini
```

Probably there's more information that's required to debug this, but I don't know what all to include here. Please tell me what to include here if something is needed.

ivankravets, 2019-01-17:
Please run `pio upgrade --dev`

closed 2019-01-17 by ivankravets

OttoWinter, 2019-01-17:
That fixed it, thanks!